kakopappa/sinric #1: German translation

x-t-i-a-n, 2018-01-05:
I am in Austria, and sadly the SKILL is not available on amazon.de.

Would you consider publishing it in German too?

I would gladly assist translating phrases to German.

kakopappa, 2018-01-05:
Thanks a lot. I did select all counties when I published the app.

I will check and get back to you

MacSass, 2018-01-06:
Hi Kakopappa,
same here - being in Germany it is not available on amazon.de skill shop and I can´t seem to install from amazon.com shop ... I´d love to test it out and provide feedback.

Great you invest your time in this - sounds like a great solution to work around the issues seen previously.

Regards - MacSass

kakopappa, 2018-01-07:
Hi @x-t-i-a-n  @MacSass 

Can you guys translate these to German, please?

Short Skill Description:
Connect your WeMos mini D1, ESP8266, ESP32 or Arduino development boards with Amazon Alexa

Full Skill Description:
With Sinric, you can connect your WeMos mini D1, ESP8266, ESP32, Arduino development boards with Alexa without emulating as a Belkin switch. 

How to use

1. Goto sinric.com
2. Create an account and log in
3. Create a new device. eg: tv
4. Use the API key displayed on the home screen to connect your development board.

Here is an example of how to connect : 
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

replace the <Place your API key here>,<your wifi ssid>, <your wifi password>

4. Install / Enable Sinric smart home skill
5. Ask Alexa to discover devices.
6. Say Alexa, turn on "tv"

Example Phrases:
Alexa, turn on tv
Alexa, turn on bedroom lights
Alexa, dim bedroom lights

Keywords (Search terms):
Connect ESP8266 to Alexa, connect wemos mini to alexa, wemos mini emulate alexa

x-t-i-a-n, 2018-01-07:
Hi Kakopappa,

glad to help.

Best Regards, xtian

.............................
Kiurze Skill Beschreibung: 

Verbinden Sie Ihre WeMos Mini D1, ESP8266, ESP32 oder Arduino Development Boards mit Amazon Alexa 

Komplette Skill Beschreibung: 

Mit Sinric können Sie Ihre WeMos Mini D1, ESP8266, ESP32, Arduino Entwicklungsboards mit Alexa verbinden, ohne einen Belkin Switch zu emulieren.

So verwenden Sie den Skill:

1 Gehen Sie zu sinric.com 
2. Erstellen Sie ein Konto und melden Sie sich an 
3. Erstellen Sie ein neues Gerät. zB: TV 
3. Verwenden Sie den auf dem Startbildschirm angezeigten API-Schlüssel, um Ihr Entwicklungsboard zu verbinden. 

Hier Beispiel für die Verbindung: https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino 

Ersetzen Sie "Place your API key here", "your wifi ssid" und "your wifi password" im Code mit Ihren Details   (***NOT TRANSLATED, CODE WILL BE IN ENGLISH***)

5. Install / Enable Sinric Smart Home Skill 
6. Bitten Sie Alexa, neue Geräte zu suchen
7. Sagen Sie Alexa, schalte "tv" ein 

Beispiele: 
Alexa, schalte TV ein 
Alexa, schalte Schlafzimmerlicht ein
Alexa, dimme Schlafzimmerlicht 

Schlüsselwörter (Suchbegriffe): 
Verbinden Sie den ESP8266 mit Alexa, steuern Sie einen wemos Mini mit Alexa

kakopappa, 2018-01-07:
Submitted to Amazon. Should be ready in a day or two. I will update this issue again when i get the published email

kocmoc-krombiera, 2018-01-19:
@kakopapoa i wanted to send you some beers - but unfortunately I can’t sign up for sinric. If you could be so kind to help me? And get you some beers? Thanks

kakopappa, 2018-01-19:
Hello,

What is the problem? Registration does not send you a confirmation email.
Simple password will do.

Let me know if you need any assistance

kocmoc-krombiera, 2018-01-19:
Hey,

Yes, I only get 404 page when trying to submit the sign-up.

So it means I can not sign up for a account.

Thanks for your help!

kakopappa, 2018-01-20:
@kocmoc-krombiera Please use Chrome. I just tried creating a test account and it worked.

NedSin, 2018-01-21:
Hey Guys,
i cant find the skill at Amazon Germany ... 

not published now or ive been to Stupid ??? 

Sorry, bad Englisch Skill ;) ...

Greetings Ned

MacSass, 2018-01-21:
@NedSin : It is not available yet, but Kakopappa is working on it and hopefully it will be available soon - I´m waiting for it as well ...

Regards - MacSass

kakopappa, 2018-01-24:
it should be available in Germany now. Can you guys check and let me know.

MacSass, 2018-01-24:
Hey Kakopappa,
I can confirm that it is available for Germany now - great. Will give it a try soon ...

Thanks for your work on this - MacSass

kakopappa, 2018-01-30:
Thanks guys, I will close this issue now

closed 2018-01-30 by kakopappa