kakopappa/sinric #2: New account already has devices?

MacSass, 2018-01-07:
Hi Kakopappa,
I created a new account on www.sinric.com but when I entered the account it already had several devices in there that look like devices from other folks? Or are those just sample devices I can delete?

Thanks - MacSass
[image]

Also I would need some more explanation how I actually can include code to switch my devices from the ESP (e.g. I want to send RF433 signals when a switch is triggered). I´m not sure where I would add that to the .ino code ... some more help would be appreciated.

kakopappa, 2018-01-07:
Well this is bad. I am on it.

kakopappa, 2018-01-07:
It's fixed now.

kakopappa, 2018-01-07:
I will update the example code with more details. that way it will be helpful for everyone

kakopappa, 2018-01-07:
example added
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

MacSass, 2018-01-07:
Hi Kakopappa,
great - sinric.com account looks good now.

You might want to add a "Edit" Button next to "Test" so a device can be edited instead of deleted and created new - although right now that would not be too complicated ...

Also the example is helpful - should be able to work with it, although I need to look deeper into what possibilities it offers (would be interesting to have the ability to set different levels and more devices like TV with e.g. "Volume up/down"). But that is for the future :-)

Thanks for your work on this - as soon as the skill is on Amazon.de I will give if a test run ...

Regards - MacSass

PS: The skill obviously needing internet is no issue in that case as Alexa needs internet available anyhow - Alexa is not for the data privacy cautious :-)

Ameengray, 2018-01-08:
Hi, the example file give an error when complied 

Arduino: 1.8.4 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

C:\Users\Ameen  Khan\Desktop\G-Plug\g-plug arch\switch_example\switch_example.ino:5:30: fatal error: WebSocketsClient.h: No such file or directory

 #include <WebSocketsClient.h>

                              ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

kakopappa, 2018-01-08:
Looks like websocket library is missing

Install

https://github.com/Links2004/arduinoWebSockets

Ameengray, 2018-01-10:
@kakopappa thanks, could also share how sinric.com code ? i want understand how the communication works, will you be adding mqtt support ? do let me know thanks. 

Great work BTW !

closed 2018-09-06 by kakopappa