kakopappa/sinric #4: Show device id

kakopappa, 2018-01-07:
Show device id of each device created in dashboard

TheKerry, 2018-01-28:
Heyho, where can I see the ID of my created Device?

kakopappa, 2018-01-28:
Right now it's not visible in the website. I am planing to fix this asap.
Sorry about that

Till then, when you invoke the device via Alexa, it is in the message you
receive.

String deviceId = json ["deviceId"];

So different devices will have different device id.

I will update this issue when it's fixed.

TheKerry, 2018-01-28:
Allright, what exactly do you mean with that?
> Till then, when you invoke the device via Alexa, it is in the message you
receive.

When I go to Smart Home / Devices I can select my Device wich I created at sinric.com, but I can't see the ID of the Device.
Or is there a voice command to recive the Device ID?

kakopappa, 2018-01-28:
Never mind, I just updated the UI to show the device id on the dashboard. 

Thanks

TheKerry, 2018-01-28:
Great, thanks. Keep up the good work! :)

closed 2018-01-30 by kakopappa