kakopappa/sinric #6: Not in Google home

MacGyver1rjq104, 2018-01-10:
I don't see this app in google home but it states it will work?  Is this a mistake?

kakopappa, 2018-01-10:
I didn't commit it yet. Still testing.

I will update the site soon

Ameengray, 2018-01-10:
Can we use the same code to control a relay using node mcu or ESP01 ?

drbeat, 2018-02-25:
Also curious if and how this may be possible

JeffreyKlassen, 2018-02-26:
Google home support planned? I do not see anything on the main site.

kakopappa, 2018-02-27:
yes, it's planned. need to fix few loose ends. I will update this issue when it's released.

JeffreyKlassen, 2018-02-27:
Sounds good. Looking forward to it.

kakopappa, 2018-03-17:
After a lot of work finally submitted to Google. fingers crossed.
 
[image]

kakopappa, 2018-03-17:
Once it get approved i will update the instructions on how to use.

saikhurana, 2018-05-14:
Any Updates???
If it is still not updated, is there any dev version we can use to connect sinric with Google Home

kakopappa, 2018-06-25:
Google home support added.
https://github.com/kakopappa/sinric/wiki/How-to-use-Google-Home

closed 2018-06-25 by kakopappa