kakopappa/sinric #8: Multiple devices on a single 8266?

DSSCircuits, 2018-01-13:
Will the code support multiple devices on the same ESP8266?  I have been using your other wemo switch software (great job by the way!!!), except it starts having connection issues when I get up to 4 device ID's.

Will this new skill support multiple device ID's? It's a little confusing trying to reference the example.  

Thanks again!!!

kakopappa, 2018-01-14:
Multiple switches will have a different device id.  Each device you create on the website has a unique device id. Unfortunately, this device id is not shown on the website. I will fix this. 

https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino#L38

Check the device id and code your logic accordingly.

kakopappa, 2018-01-28:
@DSSCircuits 

I have updated the UI to show the device id now.

closed 2018-01-28 by kakopappa