kakopappa/sinric #10: Can't control device from Alexa app, can only edit.

DSSCircuits, 2018-01-14:
So far I love the ease of use of this code!!! 
There seems to be a problem with controlling the device in the Alexa app (Android).  After discovering the device, when you click on the Icon it takes you to the Device Settings screen.  There doesn't appear to be anyway to turn on/off the device from the Smart Home section.  It happens for both the Light and Switch type devices.  Works fine using voice activation.

kakopappa, 2018-01-28:
@DSSCircuits 

Thanks :) Sorry for the late response.

Is it available for other smart-home device skills? 
I need to spend more time to check why it does not show the device settings when using a smart-home skill. Maybe it requires sending more information related to the switch.

I am also working on an Android app.  I will update this issue again when I have something.

MacSass, 2018-01-30:
HI @kakopappa ;
in order to see the on/off switches in the Alexa app I think the device needs to be able to report back the current power state.
I guess this is not reported yet - thus the App does not show the option.
See information on powerstate here:
[https://developer.amazon.com/de/docs/device-apis/alexa-powercontroller.html](url)

regards - MacSass

kakopappa, 2018-01-30:
@MacSass Thanks for checking.

At the moment, Sinric server sends the on/off result back to the Alexa server after forwarding the request to ESP.

Have you tried any other smart-home skills which does this?

DSSCircuits, 2018-01-31:
All the other smart home skills give the ability to switch on and off.  All the other skills seem to request the status of the device when you click on the device.

MacSass, 2018-01-31:
@kakopappa:
I think "just" reporting the status as result of an Alexa action request is not sufficient. Alexa also queries the status when e.g. you enter the Alexa app and as well during discovery. Thus I think the "alexa-powercontroller" powerstate needs to be implemented, so that Alexa can always verify the current status of e.g. a lamp. Then this should make the on/off available in the app as well, I guess.
I don´t have much experience with smarthome skill, but at least for the espalexa and multi-wemo ones it was available in the app ...
Regards - Macsass
PS: Maybe at the same time it would be possible to add the Alexa brightness controller interface 
[https://developer.amazon.com/de/docs/device-apis/alexa-brightnesscontroller.html](url)
or powercontroller
[https://developer.amazon.com/de/docs/device-apis/alexa-powercontroller.html](url)
which should make "dimming" capability available ...

kakopappa, 2018-02-01:
I will take a look at it on this weekend. 

@MacSass I  have already implemented 

Alexa.PowerController
Alexa.BrightnessController
Alexa.ColorController
Alexa.ColorTemperatureController

for the light switches and it should work.  I think I have to update the website to show which capabilities it can support when you select a device type. 

Please check and let me know.

MacSass, 2018-02-01:
Hi @kakopappa ,
this sounds great - unfortunately this does not seem to work yet.
I have created a new device (light) and did a new discovery from Alexa. Device shows up, but there are no options to turn it on / off or change brightness or color:

[image]

I would have expected something like this (this is a Ikea lamp). It offeres on/off, brightness and color temperature to be changed:

[image]

If I can be of any help getting that resolved - let me know (and sorry for the german screenshots).

Regards - MacSass

emnik, 2018-02-04:
@kakopappa Hi! to be able to control the device from the app, I think you should somehow need to populate the type (typ in german) field that you see in the first of the above photographs! 

When I was using the espalexa code then regarding what I set there it resulted in a different icon and different options. If I did not set the type, then it showed a kind of wifi icon and I couldn't control the device from alexa app as in the last photo.

The commands I used that resulted in different icon and options were part of the phillips hue protocoll but may help you:
`//  return "{\"type\":\"Extended color light\",\"manufacturername\":\"OpenSource\",\"swversion\":\"0.1\",\"name\":\""+ friendlyName[deviceId-1] +"\",\"uniqueid\":\""+ WiFi.macAddress() +"-"+ (deviceId+1) +"\",\"modelid\":\"LST001\",\"state\":{\"on\":"+ boolString(yourVal[deviceId-1]) +",\"bri\":"+ briForHue(yourVal[deviceId-1]) +",\"xy\":[0.00000,0.00000],\"colormode\":\"hs\",\"effect\":\"none\",\"ct\":500,\"hue\":0,\"sat\":0,\"alert\":\"none\",\"reachable\":true}}";
// return "{\"type\":\"Dimmable light\",\"swversion\":\"66011639\",\"name\":\""+ friendlyName[deviceId-1] +"\",\"uniqueid\":\""+ WiFi.macAddress() +"-"+ (deviceId+1) +"\",\"modelid\":\"LWB014\",\"state\":{\"on\":"+ boolString(yourVal[deviceId-1]) +",\"bri\":"+ briForHue(yourVal[deviceId-1]) +",\"alert\":\"none\",\"reachable\":true}}";
//  return "{\"type\":\"On/off light\",\"swversion\":\"66011639\",\"name\":\""+ friendlyName[deviceId-1] +"\",\"uniqueid\":\""+ WiFi.macAddress() +"-"+ (deviceId+1) +"\",\"modelid\":\"LWB014\",\"state\":{\"on\":"+ boolString(yourVal[deviceId-1]) +",\"alert\":\"none\",\"reachable\":true}}";`

So the type was one of:
Extended color light --> this for on/off, dim and color capabilities
Dimmable light --> this for on/off and dim only capabilities
On/off light --> just on/off capabillity

also the above types were case sensitive! 
I hope this helps a little...

MacSass, 2018-02-04:
@emnik;
This seems to make sense to me, I hope kakopappa will be able to figure it out.

I experienced that change of color and color temerature is already working from Alexa (but not the app), while change of brightness I could not get to work from Alexa (and not the app of course).

Good progress - once the small quirks have been ironed out I think we will have a very universal skill available to combine ESP with Alexa.

Regards- MacSass

kakopappa, 2018-02-10:
I have updated the site with ReportStatus . Can you guys check now ?

emnik, 2018-02-10:
Yeap, works here! I can now control me devices through the app. The icons are the right ones now and also the correct type is shown.

DSSCircuits, 2018-02-10:
Works for me as well. Great job!!

MacSass, 2018-02-11:
Hi,
I do agree - this looks already very good now - I can control on/off, brightness and color on Alexa app!
Great job @kakopappa !

On thing remaining for the icing on the cake: 
- Thermostat works well for setting a certain temperature, but has a problem with saying "Make ... warmer/colder". The response is always "XXX does not respond currently".
- Would it be possible to "feed" the thermostat status regulary to sinric.com, so it can report updated status to Alexa? Or could it query the ESP for current value when Alexa queries the temperature of a thermostat?
Currently you can ask alexa for the temperature and it responds if it was set through sincirc, but of course it would not get updated information from sinric if the temperature is changes outside of sinric/alexa.
Sinric would either have to query the ESP for current value when Alexa asks for status or there would need to be a way to regulary update sinric with the lastest value

Actually having sinric query the ESP for the current status might make sense for the light/switch as well, as it might happen, that the status is changed outside of sinric and then sinric would report a "wrong" status.

Anyway - this is getting the most useful skill for us - great job!

Regards - MacSass

kakopappa, 2018-02-12:
@MacSass Thanks for your input.

1. I will look in to this. 

2. I have already added a new api for this but i  did not publish it yet because :

I do not want people to flood the server with updates. This will delay responding to other requests. As you know Arduino's loop function can send tons of update requests to server if coded incorrectly.

Let me know what you guys think

MacSass, 2018-02-12:
Hi @kakopappa,
I do agree that it might inherit some "risk" if people "push" data to sinric.
First of all the stability of the service should have the hightest priority, as there is nothing worse than unreliable smarthome components.

So, I´m wondering if Sinric would be able to "query" the ESP for the current value when requested by Alexa. That way Sinric.com would initiate the response, not the ESP. I have implemented this right now for my own skill (not a smarthome skill), querying a PHP skript on my webpage for a temperature value ...

Also it would make sense to implement that "query" functionality to more than just thermostats, as you always need to figure in that some things get switched/changed outside of the sinric/Alexa environament.

If you want to discuss by mail, feel free to get in touch with me.

Regards - MacSass

BoriKing, 2018-02-12:
can someone please help me get this working? i'm no coder:
I already , copy api + device id from sinric.com to example.ino. but still can't get anything from alexa app to esp. only thing i can get is a test text if i press the test button in sinric.com.

i'm i suppesed to do anything here?:
    
case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
        // Example payloads

        // For Light device type
        // {"deviceId": xxxx, "action": "setPowerState", value: "ON"} // https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html
        // {"deviceId": xxxx, "action": "AdjustBrightness", value: 3} // https://developer.amazon.com/docs/device-apis/alexa-brightnesscontroller.html
        // {"deviceId": xxxx, "action": "setBrightness", value: 42} // https://developer.amazon.com/docs/device-apis/alexa-brightnesscontroller.html
        // {"deviceId": xxxx, "action": "SetColor", value: {"hue": 350.5,  "saturation": 0.7138, "brightness": 0.6501}} // https://developer.amazon.com/docs/device-apis/alexa-colorcontroller.html
        // {"deviceId": xxxx, "action": "DecreaseColorTemperature"} // https://developer.amazon.com/docs/device-apis/alexa-colortemperaturecontroller.html
        // {"deviceId": xxxx, "action": "IncreaseColorTemperature"} // https://developer.amazon.com/docs/device-apis/alexa-colortemperaturecontroller.html
        // {"deviceId": xxxx, "action": "SetColorTemperature", value: 2200} // https://developer.amazon.com/docs/device-apis/alexa-colortemperaturecontroller.html
 
any help will be appreciated.

MacSass, 2018-02-12:
Hi BoriKing,
depends on what you want the ESP to do, when a command from Alexa (via Sinric) is received.

Take a look at the switch.ino example. 
You need to paste your device ID for the switch you created on sinric to replace the 5axxxxxxxxxxxxxxxxxxx sample device ID. You need to do this in the "turnOn" and "turnOff" function.

`void turnOn(String deviceId) {
  if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
  } 
  else if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of second device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
}`

Then those functions are called when you ask Alexa to turn on/off your device.
Of course in the sample code all you get is a message on the serial console saying "Turn on device id: ...".
Here you need to put your code that tells the ESP what it actually should do, when the command is received.
What you code here will be depending on what you want the ESP to do, e.g. turn on a relay or a motor, etc.

I´m afraid you will not make a lot of use from this without at least some coding ...

Hope that helps - MacSass

BoriKing, 2018-02-12:
Thanks MacSass 
unfortunately i had already set all of this and no " Turn on device id:" shows  up on serial console. By asking Alexa or by alexa app. Only a test text from the test button when i press on it in sinric.com website.

BoriKing, 2018-02-12:
Even from the 2 devices i created in sinric.com only 1 is discovered by Alexa.

kakopappa, 2018-02-13:
@BoriKing normally it detects fine. 

do you have other devices with the same name? Goto alexa, forget all existing once and try again discovering

BoriKing, 2018-02-13:
@kakopappa
 no other devices with the same name. I even tried forgetting all my smarthime devices. No luck.

kakopappa, 2018-02-13:
@BoriKing Did you remove the other device? In the database, i can see only 1 device created using your account ?

BoriKing, 2018-02-13:
@kakopappa 
yes, i removed the other device..ill try and delete/add everything from sinric.com, disable/re-enable skill, delete everything from  Alexa/alexa app. In other words  "start from scratch " once again. Ill keep you posted.

BoriKing, 2018-02-13:
@kakopappa 

ok i was able to figure it out. 

the problem was that when you register to sinric.com the email address and password are case sensitive.
so i registered the other day to sinric.com. and entered:

example: Jhon30@gmail.com
then i tried to link the skill in alexa and entered:
example:  jhon30@gmail.com instead of Jhon30@gmail.com
and  didn't let me enable skill cause "username and password not found" so i created another account with jhon30@gmail.com meaning 2 accounts were created with the same email address "1 with lower case (j) and 1 with upper case (J).

in other words i wasn't using the right account.

kakopappa, 2018-02-14:
@BoriKing Thanks for letting me know. I have fixed this issue now.