kakopappa/sinric #9: You are not eligible to enable this skill due to geographical restrictions

IOTHobby, 2018-01-14:
Thanks for sharing Kakopappa!

Trying to get Sinric skill, getting this message
"You are not eligible to enable this skill due to geographical restrictions."

Is there is a restriction to get skill?

kakopappa, 2018-01-14:
I think it's available for US only at the moment. I am in the process of
supporting other countries. ( Germany and India)

Where are you from? Drop me your email address and I will send you an
invite(not sure that will work)

IOTHobby, 2018-01-14:
Vancouver, Canada.

kakopappa, 2018-01-14:
send me your email. I will send you an invite to beta test. That might work

IOTHobby, 2018-01-14:
rajat.soni@gmail.com

kakopappa, 2018-01-14:
I just sent you an invite. please check

IOTHobby, 2018-01-14:
Thanks a lot!! It works, was able to connect device with echo and send commands. Will try adding more nodemcu devices and start using them!

TetsudouSenpai, 2018-01-15:
Hi @kakopappa ! Would it be possible for me to also join beta testing? I'm from the UK, my email is dimaleyko@gmail.com .

Thank you very much in advance!

kakopappa, 2018-01-15:
@TetsudouSenpai sent

robsorn, 2018-01-15:
Hi @kakopappa, connecting to your service worked like a charm. However, I also can't find the skill.
Would you mind sending me a beta-testing invitation to krsstk@gmail.com (Located in Germany)?

closed 2018-01-16 by IOTHobby