kakopappa/sinric #14: Alexa doesn't seem to be communicating properly

directhacker, 2018-01-28:
Hey, want to use this to trigger a relay. I don't see where you define which pins to use. I got as far as connecting to Alexa, but when I tell Alexa to turn on or off the relay, she says "Okay" but then nothing happens. I've tried every single other way to control a relay with Alexa and an ESP8266 d1 mini, but this is the first way that I can get Alexa to discover the device

MacSass, 2018-01-30:
Hi @directhacker ,
at least you should see some output on the serial console of your ESP when Alexa says that everything is OK.
If you only use one device you can add your code to trigger the relay in those routines of the example code:

> void turnOn() {
  Serial.println("Turn on ...");
}
void turnOff() {
  Serial.println("Turn off ...");
}

Alternatively if you want to control more than one relay you can take a look at my sample code for multiple devices I posted.
As by accident posted it to a fairly unrelated thread - so I have attached it here again:
[switch_multi_sinric.txt](https://github.com/kakopappa/sinric/files/1675753/switch_multi_sinric.txt)

It has some more comments - which I hope you find helpful to understand how it works.
Of course you will have to add your own code to actually turn on / off the relay ...

Regards - MacSass

kakopappa, 2018-01-30:
This module only get the on/off request from Alexa. it's up to you to turn on or off the relay. If you are using wemos relay shield try this https://wiki.wemos.cc/products:d1_mini_shields:relay_shield

anums18, 2018-02-04:
Hi @directhacker 
I had modified the code on below link with addition of 4 channel relay.
https://gist.github.com/anums18/078e0671ba28af2a809cb9dea7a131bb

CEStanley58, 2018-02-17:
Hi @directhacker,
Mods to make the WEMOS D1 are to set the GPIO pin(s) to OUTPUT as follows:
pinMode(pinX, OUTPUT);
where pinX = D0, D1, D2, D3 ... D7, D8
Do this in the setup() function/method.
Then write the pin to HIGH or LOW:
digitalWrite(pinX, HIGH); or digitalWrite(pinX, LOW);
Do this in the tourOn() or turnOff() functions/methods.

Hope this is a direct answer to your question.

pacostello, 2018-03-18:
I appear to be having the same issue.
Wemos D1 mini.
Started with the Switch example code.
Put in my SSID, password, API code and device code.
I have defined D1 and BUILTIN_LED as outputs in void setup()
Used digitalWrite to set the above high and low in void turnOn and turnOff.
Nothing happens.
I can see on the Alexa app that it believes the device is being switched on and off but it appears to just be talking to the API not the Wemos, as it does the same regardless of whether the Wemos is powered or not!

CEStanley58, 2018-03-18:
Hi pacostello,

 

  My issue is cleared up. My Alexa thinks the WeMOS does not respond but the LEDs do show the WeMOS operating correctly.

 

Charles Stanley

kakopappa, 2018-03-19:
Hello.

1. Does  Arduino IDE console output says waiting for commands?

2. Make sure device id is correct

CEStanley58, 2018-03-19:
The Arduino IDE console is clear, no text. My device is WEMOS D1 R2 & mini. It programs correctly and runs as commanded when I select the correct board. 

 

Charles Stanley

CEStanley58, 2018-03-19:
Here is my code. Please ensure you have “Generic ESP8266 Module” or “WEMOS D1R2 & mini” selected.

Port must be set in Tools,

Load speed on my system is good at 921600 but also works at lower speeds.

 

Change the TODO’s to meet your setup.

 

Cheers,

 

/*

Version 0.1 - Feb 10 2018

https://sinric.com/login

*/ 

 

#include <Arduino.h>

#include <ESP8266WiFi.h>

#include <ESP8266WiFiMulti.h>

#include <WebSocketsClient.h> //  get it from https://github.com/Links2004/arduinoWebSockets/releases 

#include <ArduinoJson.h> // get it from https://arduinojson.org/ or install via Arduino library manager

 

ESP8266WiFiMulti WiFiMulti;

WebSocketsClient webSocket;

WiFiClient client;

 

#define MyApiKey "YOUR API KEY" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard

#define MySSID "YOUR SSID" // TODO: Change to your Wifi network SSID

#define MyWifiPassword "YOUR WiFi PASSWORD" // TODO: Change to your Wifi network password

 

#define API_ENDPOINT "http://sinric.com"

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

 

#define MASTERBEDROOM D5

#define DEN D6

 

String Device1 = "YOUR SINRIC DEVICE1" // TODO: Change to your www.simric.com device number 1

String Device2 = "YOUR SINRIC DEVICE2" // TODO: Change to your www.simric.com device number 2

[... 314 lines cut ...]
  {

      uint64_t now = millis();

      

      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass

      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) 

      {

          heartbeatTimestamp = now;

          webSocket.sendTXT("H");          

      }

  }   

}

 

Charles Stanley