kakopappa/sinric #18: Version with WifiManager

mrlightsman, 2018-02-01:
It would be great if you had a version that used a WifiManager so the network credentials didn't have to be hard coded. Makes it easier to move the device from location to location.  Thanks.

kakopappa, 2018-02-01:
Cool.

Make one and send me? I will upload it to the github

mrlightsman, 2018-02-01:
Aruna,
Thanks for the quick reply.  I've been experimenting with your Sinric
project for a couple of days now.  It is great stuff for guys like me who
lack the know how, but have the interest.

I used this GitHub project, https://github.com/tzapu/WiFiManager,  on an
earlier version of a wifi switch I experimented with.  It seems to work
pretty well .  Specifically, I prefer the AutoConnect version.

Admittedly, I don't have enough knowledge of coding to incorporate this
with your code.

On a different note, there is an instructable by samsungite called "Alexa
and Switch Operated Lamp" that is based on nassir-maliks GitHub,
https://github.com/nassir-malik/IOT-ESP8266-ESP12E-Alexa-Multiple-Devices,
(which is based on your work) that includes an interrupt button on one of
the pins to allow for manual on/off control of the switch.  He has modified
the ino, but uses all the same libraries.  (Of course, the Amazon devices
cannot discover the ESP, similarly as with your earlier project.)  This
type of added functionality would really make my wife and kids happy, as
they just don't appreciate the techiness of voice controlled environments.

I wish I could contribute, but for now, I must settle with being a leach.
Thank you for your work on this.

mrlightsman, 2018-02-01:
After reading through the tzapu GitHub, I found this variant by the real orca.  https://github.com/the-real-orca/WiFiManager  in the example folders is a version which allows for multiple networks with fs.  Assuming it works (as I have not tested it), this would really provide flexibility to device location and connection.  If I understand it correctly, it will remember former networks and connect to the strongest one if the last used network becomes unavailable.  In my case, this could be helpful for when power goes out and the ESP powers up before the AP, while still allowing for a web portal for initial use or change in location.

BoriKing, 2018-02-13:
@mrlightsman 
@kakopappa 

Added AutoConnectMultipleNetworksWithFS from https://github.com/the-real-orca/WiFiManager
and it's working wonders for me.

Remember i'm no coder i just like to read and follow instruccions.

`#include <FS.h>                   //this needs to be first, or it all crashes and burns...

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
//#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  get it from https://github.com/Links2004/arduinoWebSockets/releases 
#include <ArduinoJson.h> // get it from https://arduinojson.org/ or install via Arduino library manager
#include "WiFiManager.h"          //https://github.com/tzapu/WiFiManager

//ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "xxxx" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
//#define MySSID "xxxx" // TODO: Change to your Wifi network SSID
//#define MyWifiPassword "xxxxx" // TODO: Change to your Wifi network password

#define API_ENDPOINT "http://sinric.com"
#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void load(WiFiManager &wifiManager) {
  //clean FS, for testing
//  SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");
  if (SPIFFS.begin()) {
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file...");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial); // debug output to console
        if (json.success()) {
          Serial.println("\nparsed json");
          if ( json["Networks"].is<JsonArray>() ) {
            for (int i=0; i < json["Networks"].size(); i++) {
              auto obj = json["Networks"][i];

              // add existing networks and credentials
              wifiManager.addAP(obj["SSID"], obj["Password"]);
            }
          }
        } else {
          Serial.println("failed to load json config");
          return;
        }
      }
    }
[... 166 lines cut ...]

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}`

BoriKing, 2018-02-13:
@mrlightsman 
@kakopappa 

Added AutoConnectMultipleNetworksWithFS from https://github.com/the-real-orca/WiFiManager
and it's working wonders for me.

Remember i'm no coder i just like to read and follow instruccions.

[https://gist.github.com/BoriKing/2261be246463680e7de1ec7452dfb6ab](url)

kakopappa, 2018-02-14:
@BoriKing Thanks. https://github.com/kakopappa/sinric/blob/master/arduino_examples/multi_wifi_example.ino added

mrlightsman, 2018-02-15:
@boriking

First, thanks for the code update. This will make the chips very user friendly. 
But, I keep getting an error I can’t figure out when programming the chip with Arduino ide.  It tells me there is no getap or addap in wifimanager. (Sorry, I’m not at my computer and can’t remember the exact error text.)
It seems I must have an old wifimanager library? But I’ve used the tzupa (sp?) one in the compilation. I’ve been using his older version of the Wi-Fi manager for about a year on another project. But when I pull the real orca files I get nothing....

Thoughts and help are greatly appreciated. Thanks.

BoriKing, 2018-02-15:
@mrlightsman 

for Multi WiFi you need modified version of   WiFiManager.cpp and WiFiManager.h from https://github.com/the-real-orca/WiFiManager 

for the autocconect you can still use tzupa.

mrlightsman, 2018-02-15:
Thanks. That must be my issue. I’ll give it a go later.

mrlightsman, 2018-02-16:
@boriking
I updated my Library and everything compiles as expected. When I load the chip and try to run the code it errors. The serial monitor shows it crashing after passing the json file. It says there is an error 28 then loops a bunch of garbage before rebooting. Seems to have something with the strcpy code. 
I’ll try to post some screen shots later. Thanks.

BoriKing, 2018-02-16:
@mrlightsman you must reset your device as it's reading your old configurations i will be adding that in a few.

BoriKing, 2018-02-17:
@mrlightsman flash this: [Format Device](https://github.com/BoriKing/WiFi-Manager/blob/master/Format%20Device)

Then flash the sketch you were trying.

mrlightsman, 2018-02-17:
@BoriKing 
Thank you for the help with flashing the esp.  I did that, then immediately loaded the wifimanager code of yours.  Here is what the serial is reporting now.  (it should be noted, when I download your sketch, Arduino ide can't open it. So I've had to open it as a text doc and paste it into Arduino.  Any idea what might be going on?  I don't see anything that I can make sense of.  Thanks.

mounting FS...
mounted file system
*WM: Adding parameter
*WM: api key
*WM: Adding parameter
*WM: 1st-deviceId
*WM: Adding parameter
*WM: 2nd-deviceId
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.0.19
connected...yeey :)
local ip
192.168.0.19
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
[WSc] Webservice disconnected from sinric.com!
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
[WSc] Webservice disconnected from sinric.com!
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...

The last few lines go endlessly and I cannot get a response from alexa..

BoriKing, 2018-02-17:
@kakopappa 

can you comment this: //https://github.com/the-real-orca/WiFiManager
instead of: //https://github.com/tzapu/WiFiManager
next to #include "WiFiManager.h"   
to the [multi_wifi_example.ino](https://github.com/kakopappa/sinric/blob/master/arduino_examples/multi_wifi_example.ino)

and can you add  wifiManager.autoConnect("Sinric"); right above 

  if(!wifiManager.autoConnect()) {
    Serial.println("failed to connect and hit timeout");
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(1000);

so that it creates  n AP named Sinric instead of Espxxxxxx 
thanks.

BoriKing, 2018-02-17:
@mrlightsman 

flash this: [FormatDevice.ino](https://github.com/BoriKing/WiFi-Manager/blob/master/FormatDevice.ino)
once more: added WiFi manager reset.

mrlightsman, 2018-02-17:
@BoriKing 
YES. That did the trick.  Thank you.  I am now going to experiment with making pin 3 (RX) the trigerpin and add some feedback on pin 1 (TX) to use the built in LED as an indicator.  I'll let you know how it goes.

mrlightsman, 2018-02-17:
@BoriKing 
Things are working excellent now.  THANK YOU for helping me through this project.  I've added code to give a quick double flash when connected to WiFi.  Then flash 1 for device 1, flash 2 for device 2, flash 3 for unknown device.  Next I will try to find a spot to cause a steady blink while connecting to WiFi (versus connected), but I'm not seeing an obvious spot to add that.

The LED feedback is helpful to ensure the device is reacting to commands as expected.  I am doing that on pin 1 and using pin 3 as the reset pin.

mrlightsman, 2018-02-17:
@boriking and @kakopappa rock!  This is a great way to access Alexa. Thank you both for the help. It’s time for me to close the issue.

BoriKing, 2018-02-17:
@mrlightsman 

thanks for your feedback
glad i could help, for future reference the [WiFi Manager AutoConnect](https://github.com/BoriKing/WiFi-Manager/blob/master/Sinric-AutoConnect-api_key-DeviceID-Input-physical_credentials_reset.ino)

has a physical reset if you connect GPIO 2 to ground, so that way you don't have to flash [FormatDevice.ino](https://github.com/BoriKing/WiFi-Manager/blob/master/FormatDevice.ino).

BoriKing, 2018-02-17:
@mrlightsman

Here you go:
 [SinricAutoConnectCustomParametersFS_CredentialsReset.ino](https://github.com/BoriKing/WiFi-Manager/blob/master/SinricAutoConnectCustomParametersFS_CredentialsReset.ino)

This sketch will make your device led blink while in configuration mode, deleted external reset switch and added Reset credential by pressing the built in reset switch twice in less than 2 seconds.

kakopappa, 2018-02-18:
@BoriKing can you send me the updated multi_wifi_example.ino ? Thanks

BoriKing, 2018-02-19:
@kakopappa 

Here you go: [SinricMultiWiFiCustomParametersFS_CredentialsReset.ino](https://github.com/BoriKing/WiFi-Manager/blob/master/SinricMultiWiFiCustomParametersFS_CredentialsReset.ino)
added the ability to add api key + devices id while in configuration mode. Deleted hard coding api key + devices id  inside scketch + added Reset all credentials by pressing the built in reset switch twice.

mrlightsman, 2018-02-20:
@BoriKing 
I was using your earlier version with digitalRead(triger_pin, LOW) but something must be flakey with my physical build because I cannot seem to avoid pin 3 (the RX pin which has been defined as triger_pin) from being read as low, even with a 1k pull-up resistor.  Every time the esp boots, it connects to wifi, then goes right into reset.
I tried nesting two if statements like this:
If (digitalRead(trigger_pin,LOW) {
  delay(1000);
  if(digitalRead(trigger_pin,LOW) {
  do the reset code...
  }
}
My thought being that in order to reset, one would have to hold the external reset button (a NO button connected to ground) for at least a second.  That way, any flakiness might be overlooked.  But I must be nesting the If statements wrong, or something, because it doesn't work.

Now I've been working with your version with Ticker and drd.  Ticker works great.  However, the ESP-01 doesn't have an onboard reset.  Is there a way to redirect this to pin 3 (RX)?  And, might this help address my flakiness?

I cannot figure out why things are flaky with my current setup.  All connections are carefully soldered.  I've used a pull-up resistor to keep the pin from floating.  Yet, it still flakes to LOW and resets.

Any help is appreciated.  Thanks.

BoriKing, 2018-02-20:
@mrlightsman 

I will be working on that  sometime today after work..

mrlightsman, 2018-02-20:
Thanks.  I also found, late last night that changing pinMode (Triger_Pin,
INPUT) to pinMode (trigger_pin, INPUT_PULLUP) has significantly stabilized
it.  Since doing that, things have been behaving properly.

victorantonio21, 2018-09-10:
Hi, I got an Different error trying to compile the code. Can you help me solve it?
I'm using Arduino IDE 1.8.6, the Sinric Multi_Wifi_Example with the Real-Orca Wifi-Manager. the error says:

libraries\WiFiManager-WiFiMulti\WiFiManager.cpp.o: In function `~_Vector_base':

C:\Users\Documents\Arduino\libraries\WiFiManager-WiFiMulti/WiFiManager.cpp:2176: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

libraries\WiFiManager-WiFiMulti\WiFiManager.cpp.o: In function `__gnu_cxx::new_allocator<unsigned char>::deallocate(unsigned char*, unsigned int)':

C:\Users\Documents\Arduino\libraries\WiFiManager-WiFiMulti/WiFiManager.cpp:2176: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling to NodeMCU 1.0 (ESP-12E Module)

I appreciate any help <3

mrlightsman, 2018-09-10:
> `__gnu_cxx::new_allocator::deallocate(unsigned char*, unsigned int)':

Hi. I can't be of much help, as I am not much more than a newbie.  However,
when I was doing this project, I was getting errors because I had the Tzapu
WifiManager library installed.  The-Real-Orca Wifi manager had some
modifications.  Ultimately, I stayed with tzapu and did not use the
multiWiFi aspects.

Sorry I could not be more help.  Good luck.

betterligne, 2019-01-20:
> Hi, I got an Different error trying to compile the code. Can you help me solve it?
> I'm using Arduino IDE 1.8.6, the Sinric Multi_Wifi_Example with the Real-Orca Wifi-Manager. the error says: [...]

same problem. any help please

betterligne, 2019-01-20:
@victorantonio21 

problem solved. Just install this : **esp8266** by **ESP8266 Community** version **2.4.1** from boards manager

arduino-nour, 2019-08-19:
**Had a great experience with ESP8266**

crazytacowizerd, 2020-05-04:
Hi, I'm getting the same 'class WiFiManager' has no member named 'addAP' error, I have changed to the modified wifi manager which didn't work can anybody help, please?

mrlightsman, 2020-05-04:
Sorry, I’ve long since abandoned that project. What I recall, though, is
that you have to use the real orca version of Wi-Fi manager in order to use
the addap function.

crazytacowizerd, 2020-05-04:
I have added the real orca version to my libraries or is there something else I had to do?

mrlightsman, 2020-05-04:
Try uninstalling other versions of Wi-Fi manager?  Beyond that, I’m at a
loss. Sorry. And good luck.

crazytacowizerd, 2020-05-04:
Okay, i will try that thank you. if you don't mind me asking why did you abandon the project?

mrlightsman, 2020-05-04:
I didn’t like the blocking effect of Wi-Fi manager so I ended up making my
own web interface that allows web server to remain on. It is not good code,
but it does the job. This way, if I lose internet connection I could let
the rest of the program run anyway.

Wi-Fi manager is the way to go if it does what you want .

crazytacowizerd, 2020-05-04:
oh okay, well if it works for you then that's great!  just one more thing sorry.
I think its fixed my problem but I'm getting  error compiling now 
`In file included from C:\Users\luket\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/ctype.h:7:0,

                 from C:\Users\luket\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0/tools/sdk/libc/xtensa-lx106-elf/include/ctype.h:5,

                 from c:\users\luket\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\cctype:42,

                 from c:\users\luket\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\bits\localefwd.h:42,

                 from c:\users\luket\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\string:43,

                 from c:\users\luket\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\stdexcept:39,

                 from c:\users\luket\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\array:38,

                 from c:\users\luket\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\tuple:39,

                 from c:\users\luket\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\functional:55,

                 from c:\users\luket\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\memory:79,

                 from C:\Users\luket\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\cores\esp8266/FS.h:24,

                 from C:\Users\luket\Desktop\sinric-master\arduino_examples\multi_wifi_example\multi_wifi_example.ino:3:

C:\Users\luket\Documents\Arduino\libraries\WiFiManager-WiFiMulti/strings_en.h:20:36: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol

 const char HTTP_HEAD[]             PROGMEM = "<!DOCTYPE html><html lang='en'><head><meta name='format-detection' content='telephone=no'><meta name='viewport' content='width=device-width, initial-scale=1, user-scalable=no'/><title>{v}</title>";

                                    ^

In file included from C:\Users\luket\Desktop\sinric-master\arduino_examples\multi_wifi_example\multi_wifi_example.ino:7:0:

C:\Users\luket\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.0\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };`
if you could help me fix this problem too it would be great! thanks

mrlightsman, 2020-05-04:
> const char HTTP_HEAD[] PROGMEM = "<title>{v}</title>";

Sorry. That is way over my head. I’m a simple hobbyist who, like you asked
a lot of questions. I wish I could have helped more.

If you are lucky, aratuna will chime in. That guy is very smart. Good
luck.

crazytacowizerd, 2020-05-04:
That's okay I'm sure ill find my answer somewhere thank you for your help anyway :)

shahibul, 2020-05-05:
Hi, I am not able to execute the program can someone help me to execute it

SudhaMaanasa, 2020-05-30:
> @mrlightsman
> 
> Flash this: [FormatDevice.ino](https://github.com/BoriKing/Tests/blob/master/tests/FormatDevice.ino)
> 
> then new sketch.

@BoriKing 
Could you Please send this ino file to me.. I m facing the same issue

SudhaMaanasa, 2020-05-30:
@BoriKing
Also could u please send me this  SinricMultiWiFiCustomParametersFS_CredentialsReset.ino I am unable to access both of these links.