kakopappa/sinric #20: IS dimming supported yet ? {"deviceId": xxxx, "action": "", value: 0}

emnik, 2018-02-04:
Hi! have you implemented dimming?
I ask because although I've successfully used your example for openning and closing a relay, I have problem with dimming.
When I ask alexa to set device to xx (percent) the response in the serial is 
`{"deviceId": xxxx, "action": "", value: 0}`
so she doen't even recognises the action... so nothing really happens. She says "I don't know what went wrong"...

kakopappa, 2018-02-05:
Thanks for reporting. 

I have implemented dimming already but there seems to be a issue somewhere.  MacSass also said it does not work. I will look into this ASAP.

kakopappa, 2018-02-05:
@emnik fixed it. Alexa should respond with OK now.

emnik, 2018-02-05:
@kakopappa Sorry but although Alexa responds OK it sends:
[WSc] get text: {"deviceId":"5a76bebe64f318xxxxxxxxxx","action":"","value":0}
so still no action and no value...

kakopappa, 2018-02-06:
this is strage. 

I tested it and it worked.  @MacSass also confirmed me it's working. I will add more logs and check what's  going on.

MacSass, 2018-02-06:
Hi,
I did not yet actally test the ESP side - I can only confirm that Alexa responds "OK" but have not checked yet what is reaching the ESP. I will try and do so later tonight and let you know what I find.

Regards - MacSass

kakopappa, 2018-02-06:
[image]

Seems to be fine for me. How about you guys ?

MacSass, 2018-02-06:
Hi @kakopappa, @emnik ,
now I did some tets with an ESP actually connected - and I think you are both right :-)

_Increasing / Decreasing brightness works fine for me:_
... "action":"AdjustBrightness","value":25}
... "action":"AdjustBrightness","value":-25}

**Setting the brightness to a certain % level does not seem to work, although Alexa responds with OK**:
... "action":"","value":0}

_Setting a certain color works fine:_
.... "action":"SetColor","value":{"hue":240,"saturation":1,"brightness":1}}
... "value":{"hue":60,"saturation":1,"brightness":1}}

_Setting a certain light temperature works fine:_
... "action":"SetColorTemperature","value":2200}

**Increasing / Decreasing light temperature (make xxx warmer) does not seem to work**, Alexa responds with a "I don´t know what went wrong.
... "action":"DecreaseColorTemperature","value":0}

So, good progress - some thing work, some smaller quirks with some settings still ...

Regards - MacSass

MacSass, 2018-02-07:
Hi @kakopappa,
I did all my tests using Alexa as the app does not offer brightness setting or any other setting yet.

On light temperature: If Alexa does not offer a value there that is OK, then it still is helpful to have the action for completeness - so I would opt to keep it like it is, just the issue of Alexa responding with "something went wrong" needs to be fixed obviously.

Thanks for your work on this - MacSass

kakopappa, 2018-02-10:
@emnik @MacSass 

I have fixed the issues and updated the site with ReportStatus and Thermostat features. Can you guys check now ?

emnik, 2018-02-10:
it fixed "Can't control device from Alexa app, can only edit. #10" but not dimming. 
Also one more clue. 
When I use the app to change the dim value (now that the app works :-) ) it reports in the app **"the device is unresponsive"**.  So perhaps an error between the messages exchanged?
If i press on or off it works without the message. In the serial it emits the same message as previous: 
[WSc] get text: {"deviceId":"5a76bebe64f318xxxxxxxxxx","action":"","value":0}

kakopappa, 2018-02-14:
@emnik Hello, do you still have this issue? I have made lot of improvements lately please check and let me know.

emnik, 2018-02-15:
```
Dim livingroom light 20
[WSc] get text: {"deviceId":"5a7899d37d2f34455915b1bd","action":"AdjustBrightness","value":-20}

Dim livingroom light 20%
[WSc] get text: {"deviceId":"5a7899d37d2f34455915b1bd","action":"AdjustBrightness","value":-20}

Dim livingroom light minus 20
[WSc] get text: {"deviceId":"5a7899d37d2f34455915b1bd","action":"AdjustBrightness","value":-20}

-------------------------------------------------------------------------------
Increase livingroom light 50
[WSc] get text: {"deviceId":"5a7899d37d2f34455915b1bd","action":"AdjustBrightness","value":50}
-------------------------------------------------------------------------------
Decrease livingroom light 10
[WSc] get text: {"deviceId":"5a7899d37d2f34455915b1bd","action":"AdjustBrightness","value":-10}
-------------------------------------------------------------------------------
```
**So when the command is "dim" the resulting value is always negative... 
Also when adjusting the brightness the brightness level in the app doesn't change**

**Regarding the "set" command still the same behavior:**

```
Set livingroom light to 20%
[WSc] get text: {"deviceId":"5a7899d37d2f34455915b1bd","action":"","value":0}

Set livingroom light to 20
[WSc] get text: {"deviceId":"5a7899d37d2f34455915b1bd","action":"","value":0}

Set livingroom light 20
[WSc] get text: {"deviceId":"5a7899d37d2f34455915b1bd","action":"","value":0}
```

kakopappa, 2018-02-17:
@emnik fixed now. 

Dim livingroom light 20

mutthunaveen, 2018-02-17:
@kakopappa 

@emnik fixed now??.. sorry bro im still receiving "-25" when i say DIM..     :)

thanks for the code correction bro.

emnik, 2018-02-18:
@kakopappa Great improvement!
Set ... to x% is working perfectly! Dim, increase, decrease sometimes result in AdjustBrightness (as they should)  and some times in SetBrightntness...
Also the brightness status in the app is not changing. These two maybe correlated?
That said, I think the set command was the most usefull to set brightness and is now working! Bravo!

closed 2018-03-04 by kakopappa