kakopappa/sinric #22: No Value payload in"SetColorTemperture"

PSUGuy, 2018-02-12:
When I tell Alexa (using the app) to set the color temperature (not increase/decrease), there is no Value payload, has this gone?
when I ask alexa, it  (she?) tells me my value is out of range.

Excellent Work on this project by the way!

kakopappa, 2018-02-13:
Hello,  Thanks for reporting this issue. I will take a look and update the issue again when it's fixed

kakopappa, 2018-02-13:
@PSUGuy I checked the code and value parameter was missing in the payload. I fixed it and updated the server. Please check now. 

value is an integer ranging from 1000 to 10000, other than that, Alexa will tell you it's out of range.

 https://developer.amazon.com/docs/device-apis/alexa-colortemperaturecontroller.html

PSUGuy, 2018-02-13:
@kakopappa Looks like that Fixed it. Thanks.
Interestingly I can tell Alex to set to a shade of white (E.g. "warm white") but I cant say "set color temperature to 3000". I just get a value out of range response.

I'm guessing this is a limitation of the Alexa API

kakopappa, 2018-02-14:
i will close this issue now

closed 2018-02-14 by kakopappa