kakopappa/sinric #24: Sinric Down?

senic35, 2018-02-14:
Is Sinric down? I cant login to my account anymore.. it always comes up with `User not found` . I can create a new account but the problem persists. Any info on this?

senic35, 2018-02-14:
Alright i took a different E-Mail and it worked. I was able to create a new account with the same E-Mail I used before tho. Also saw issue #7

kakopappa, 2018-02-15:
There was a problem related to username being case-sentive and I fixed it and forsome reason issue still persists. 

Please use lowercase email address when you sign-up if you have still have any issues

https://github.com/kakopappa/sinric/issues/23

kakopappa, 2018-02-16:
@senic35 
I will close this issue since #23 is fixed. now. 

Let me know if you still have any issues

closed 2018-02-16 by kakopappa