kakopappa/sinric #25: Code help

Dan-Asp, 2018-02-15:
Hello All, I'm Dan. I am fairly new to coding but have a little experience with arduino. I have managed to add code to the sinric multi switch example and control 4 relays. What i would like to do is build a central heating thermostat using sinric. Any help with the code or even just a push in the right direction would be really helpful. Sorry this is not an issue with sinric and just an issue with my lack of coding experience.

kakopappa, 2018-02-15:
@Dan-Asp I will upload a example when it's ready.

kakopappa, 2018-02-17:
@Dan-Asp here you go

https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example.ino

Dan-Asp, 2018-02-17:
@kakopappa You are an international superstar!!!

kakopappa, 2018-02-18:
no you are. thanks for the donation

closed 2018-02-18 by kakopappa

Synco22, 2018-06-25:
Hello,

I was looking for such project, thank you very much. I can't find the type of thermal sensor and where to connect it. Can you please tell me?

Best regards,
Keep up the good work,
Bye