kakopappa/sinric #26: Can't login - User not found

CEStanley58, 2018-02-15:
I register, after I return to the main screen, Registration successful at top. But when I try to login, "User not found". I.E. does not show page. Firefox seams to work but cant get logged in.

kakopappa, 2018-02-15:
Hi can you try with Chrome?

kakopappa, 2018-02-15:
Sorry I should have been more clear earlier than just asking you to use
Chrome.

there was a problem with username being case sensitive and I fixed it,
tested it and all worked fine on Chrome that's why I asked you to try on
Chrome.

I will test the signup again when I get back home from work. Sorry about
that.

CEStanley58, 2018-02-15:
I don't have Chrome loaded. Chrome takes over the system.

My password is quite difficult, 20 characters.

CEStanley58, 2018-02-15:
Okay, I enabled Chrome on my mobile. Same thing, register as CEStanley58... and I get Registration successful at the top of the login screen. So I login at the bottom and the top shows User not found.

kakopappa, 2018-02-15:
I did a quick test. When you sign up, if your email address has uppercase letters there is a problem logging in. 

If you eneter the email address in lowercase, you can sign-up and login. 

Sorry about this. I will fix this when I get back home. that will be around 10 hours from now

kakopappa, 2018-02-15:
Fixed. Email address is not case-sensitive any more

CEStanley58, 2018-02-15:
I am using proper case in both user name and password. What is the maximum length of password? I had $ in my password, removed them and added more characters. 25 characters at present. Can "$#._" be used in password?

kakopappa, 2018-02-15:
No restrictions for the password.

I recommend you to create a new account. Email address does not have to be
a real one. It does not send a confirmation email. So fake one is fine.
 when you install the skill and link  you must enter the same email and
password

Hope this helps

CEStanley58, 2018-02-15:
I was able to register my email address but can't login with my password. So the password length must be an issue and not the same between the registration page and login page.

CEStanley58, 2018-02-15:
Okay, on your advise, I changed to a junk email, not mine. I now have an account. Something in gmail.com?

Anyway, you can close this issue.

Thank you for dealing with me so fast. I was not expecting to get this far today.

closed 2018-02-15 by CEStanley58

CEStanley58, 2018-02-16:
Hi Kakopappa,

 

  I changed the email address and password to all characters, no “$” and not gmail. I am now able to use your server. Great work!

 

Charles Stanley