kakopappa/sinric #27: Web or android app with sinric?

mutthunaveen, 2018-02-16:
Love your work bro... I have used wemo multiple switches worked well... Upto 4 switches i tried and it worked well...

Now im gonna try sinric this weekend...and before that i ve a question....
Along with alexa is it possible to perform on off dim etc functions using an app??

Any future plans??

kakopappa, 2018-02-16:
Thanks,

dimming is already supported. Please check for the light example in the
GitHub repo

mutthunaveen, 2018-02-17:
Bro... Not with alexa app...
With some other third-party app??

Like mqtt dash etc?? Or any other home automation app.?

kakopappa, 2018-02-17:
Alexa app should work according to  https://github.com/kakopappa/sinric/issues/10. 

Alexa app depends on reportstate https://developer.amazon.com/docs/device-apis/alexa-interface.html#reportstate and I have already added reportstate support.

I do have plans to build an Android app for this however, this may take some time because I am trying to put more time to improve the skill and website first

mutthunaveen, 2018-02-17:
Hello Dear

It is finally working good, i ve tried with multiple wifi and 2 sw and 1 lamp... 

switches are working good. but lamp feedback need improvement

Here is what i ve identified
 -> i turned ON a switch via app and after i commanded echo to turn it off -> status on alexa app updates good for switch
-> the same way, i setBrightness to 50 in app and after i commanded echo to set it to 45, in this case the printf with value 45 is coming from ESP. But, the feedback in alexa app in my phone says lamp unresponsive and value stays at 50.
-> lamp brightness auto update in app is not happening when commanded through echo device.

when u have time please have a look at it??

thanks mate.