kakopappa/sinric #28: WeMOS is working but Alexa says Unresponsive

CEStanley58, 2018-02-16:
My WeMOS device is operating correctly but the Alexa app is not receiving feedback or states the device is Unresponsive.

kakopappa, 2018-02-17:
Not sure what your current setup.

I just tested the skill and all work fine. So,

1. Unlink the skill in your Alexa app,
2. Link the skill again with the same account you used to create devices in sinric.com
3. Make sure you have the correct api key specified in Arduino example.
4. Scan for devices.

amitabh-srivastav, 2019-01-02:
@kakopappa Today I faced same problem for all my Sinric devices. The solution suggested by you worked (only steps 1, 2 and 4 are required) and the problem got solved. All devices updated their status on Alexa App and are working fine. You may treat the issue as closed. Thanks.

CEStanley58, 2019-01-02:
What are you talking about?  I am not interested.  Please remove me from your contacts.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/kakopappa/sinric/issues/28#issuecomment-450771757",
"url": "https://github.com/kakopappa/sinric/issues/28#issuecomment-450771757",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

amitabh-srivastav, 2019-01-02:
> ","avatar_image_url":"

Dear Stanley
You lack netiquette and need to be polite.
You received the email because you had raised following issue on
Github/sinric in Feb 2018, but despite resolution of issue, the thread was
not closed by you :
*"My WeMOS device is operating correctly but the Alexa app is not receiving
feedback or states the device is Unresponsive. " *
You should remove yourself from the thread on github.
regards
Amitabh

CEStanley58, 2019-01-03:
Remove me from your contact. I will block you.