kakopappa/sinric #29: multi_nav_wifi:51: error: 'class WiFiManager' has no member named 'addAP'

mutthunaveen, 2018-02-17:
Hi

I ve copied the code from "multi_wifi_example.ino"
and added all essential libraries, after compiling im getting this error

`

	WARNING: Spurious .github folder in 'WiFiManager' library
	G:\003 Projects\ESP8266\arduino\03 sinric\multiwifi_sw\multi_nav_wifi\multi_nav_wifi.ino: In function 'void load(WiFiManager&)':

	multi_nav_wifi:51: error: 'class WiFiManager' has no member named 'addAP'

				   wifiManager.addAP(obj["SSID"], obj["Password"]);

							   ^

	G:\003 Projects\ESP8266\arduino\03 sinric\multiwifi_sw\multi_nav_wifi\multi_nav_wifi.ino: In function 'void save(WiFiManager&)':

	multi_nav_wifi:76: error: 'class WiFiManager' has no member named 'getAP'

		 for (int i = 0; auto ap = wifiManager.getAP(i); i++ ) {

											   ^

	exit status 1
	'class WiFiManager' has no member named 'addAP'

`

can you tell me what is wrong??

thanks..

BoriKing, 2018-02-17:
@mutthunaveen 

for Multi WiFi, you need the modified version of WiFiManager.cpp and WiFiManager.h from https://github.com/the-real-orca/WiFiManager

see https://github.com/kakopappa/sinric/issues/18

mrlightsman, 2018-02-17:
@mutthunaveen 
@BoriKing 
Yes. This fixed my problem immediately the other day.

mutthunaveen, 2018-02-17:
@BoriKing 

thanks.. it worked..

closed 2018-03-10 by kakopappa

SudhaMaanasa, 2020-06-30:
Hi

I 've copied the code from "Sinric_Multi_WiFi_Networks_and_CredentialsReset.ino"
and added all the required libraries, after compiling I'm getting the following error

E:\Documents\Arduino\libraries\WiFiManager-WiFiMulti/strings_en.h:20:36: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol
 const char HTTP_HEAD[]             PROGMEM = "<!DOCTYPE html><html lang='en'><head><meta name='format-detection' content='telephone=no'><meta name='viewport' content='width=device-width, initial-scale=1, user-scalable=no'/><title>{v}</title>";
                                    ^
In file included from E:\Documents\Arduino\Home Automation final\Sinric_Multi_WiFi_Networks_and_CredentialsReset_GitHub_Test.ino\Sinric_Multi_WiFi_Networks_and_CredentialsReset_GitHub_Test.ino.ino:5:0:
E:\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.1\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:39: error: previous declaration of 'HTTPMethod HTTP_HEAD'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                                       ^
exit status 1
Error compiling for board Generic ESP8266 Module.

Please help me in solving this...