kakopappa/sinric #30: Sinric.com not showing my info

mrlightsman, 2018-02-17:
@kakopappa 
I noticed yesterday when I log into sinric.com my devices are not there and the API is not mine.  However, it welcomes me by name and all of my previously installed devices continue to work with alexa as expected.  Thanks.

kakopappa, 2018-02-17:
Did you have another account with the same email address (upper case/
lowecas) differently

mrlightsman, 2018-02-17:
No. Just the one account. Set up about a week ago. 

Jim

kakopappa, 2018-02-17:
Strange. Can you send me your username

mrlightsman, 2018-02-17:
Yes. Is there a way to private message so I am not sharing with everyone?

Jim

kakopappa, 2018-02-17:
Send it to aruna.tennakoon@gmail.com

Cheers

BoriKing, 2018-02-17:
@kakopappa 

for some reason after you fixed the email issue, the sinric skill was still recognizing the account that was deleted on sinric.com, to fix that i had to disable and re-enable the skill.

kakopappa, 2018-02-18:
@mrlightsman problem was you have registered for two accounts. 

1. jim@xxxx.com
2. Jim@xxxx.com

With the new update, it's doing a case-insensitve checking when it's looking for your email. Let me know which one you DO NOT USE. i will remove it.

mrlightsman, 2018-02-18:
Using either Jim or jim both take me to an account with only "Office Light" in it.  This is the wrong account.  I need access to the account that has 8-10 devices in it.  Thank you.

kakopappa, 2018-02-19:
@mrlightsman that is stragne. 

Can you email me a device id or the previous api key that you used to have? i will look it up on the database.

closed 2018-03-08 by mrlightsman