kakopappa/sinric #31: Support for rolling shutters possible?

JanB97, 2018-02-23:
Hi,
I have motorized shutters in my house.
They come with a remote which has the ability to move them up, down and to a specified (favourite) position.
Is it possible to add support for this case? (Voice command "up", "down" and something for the favourite position?)

kakopappa, 2018-02-24:
Hello, 

I think that is possible using "SkipChannels".
https://developer.amazon.com/docs/device-apis/alexa-channelcontroller.html

Alexa, channel down on **device**
Alexa, channel up on **device**

Is this the correct way to do this?

I will add a new device type called "Shutters" and update the web interface.

kakopappa, 2018-02-24:
Hello 
I have updated the website with channel controller support.

1. Create a new device. select window shutters
2. here is the sample code https://github.com/kakopappa/sinric/blob/master/arduino_examples/window_shutters_example.ino

kakopappa, 2018-02-24:
let me know if you need any changes

JanB97, 2018-02-24:
The german translation is not really ideal for this, because it's translated as "next" and "back" but that's not your fault.
Other than that, it works really good as the rest of your Project, thanks a lot!  😃

kakopappa, 2018-02-24:
Lol

kawrom, 2018-02-26:
Hi,
very good job, so far.
Is there a possibility to add commands like:
„Alexa, set [device name] to [value] %." for window shutters.
That would be exactly what I need for my SmartHome project.
Btw: window shutter from Sinric.com is not recognized as window shutter on alexa app, but as "custom controll" or "other". So its not possible to change values in the app.
thx

kakopappa, 2018-02-27:
@kawrom Thank you for your input.

1. I will add percentagecontroller support to window shutter device type soon;  https://developer.amazon.com/docs/device-apis/alexa-percentagecontroller.html

2. Right now "window shutter" device type sends "OTHER" category to Alexa. That is because I could not find the right device type here:
https://developer.amazon.com/docs/device-apis/alexa-discovery.html#display-categories

what do you think the right one to use ?

kakopappa, 2018-02-27:
OK I have added Alexa.PercentageController support for  "window shutter" device type. However, there is a small issue associated with  Alexa.PercentageController.

I send the correct response to Amazon but Alexa says "device not responding" I am not sure why.  I think this is their api issue. I see someone else also has the same problem

https://stackoverflow.com/questions/48661954/alexa-increasing-and-decreasing-brightness-of-device

kawrom, 2018-03-01:
It seems to work correctly. As you posted: Alexa always says "device not responding" - but the device is receiving the correct JSON-String. I think, I can live with it.
Thank you.

kakopappa, 2018-03-04:
I am still talking with guys at amazon to get this resolved. I will close this issue when it's fixed.