kakopappa/sinric #33: Control via physical push button when sinric server down or net disconnected ?

nilavachowdhury, 2018-03-03:
Is there any way to control the devices via physical push button in case of situations where net may be disconnected or sinric servers are down?

kakopappa, 2018-03-04:
I think you have to add a push button and do that manually. 

https://www.arduino.cc/en/tutorial/pushbutton

BoriKing, 2018-03-04:
@nilavachowdhury 

[Sinric_Multi_WiFi_Networks_CredentialsReset_and_PhysicalButton.ino](https://github.com/BoriKing/Sinric-WiFi-Manager/blob/master/Arduino%20ESP-8266%20examples/Light/Single%20Light/With%20Physical%20Button/Sinric_Multi_WiFi_Networks_CredentialsReset_and_PhysicalButton.ino)

mrlightsman, 2018-03-08:
Yes.  This can be done.  I am doing it by connecting a pushbutton to GPIO_2 to control GPIO_0.  I use digitalRead(), although there are easier ways to do it.

int GPIO_2 = 2;
Int GPIO_0 = 0;
int GPIO_2State; //State of GPIO_2 pin pushbutton
int GPIO_0State; //State of first_device

GPIO_2State = digitalRead(GPIO_2); //Toggle button for first_deviceId (LOW is pressed)
GPIO_0State = digitalReady(GPIO_0); //Is light on or off?

//If GPIO_2 is LOW toggle the first_deviceId
 if (GPIO_2State == LOW) {
//      digitalWrite(GPIO_0, !GPIO_0State);
//      delay(1000);
//    }

closed 2018-03-15 by nilavachowdhury