kakopappa/sinric #37: please help running Sinric on Raspberry pi

malvar0929, 2018-03-03:
Hi Kakopappa, could please provide me a quick guide on how I can get Sinric running on a raspberry pi.
I see the node-js sample but I'm not understanding how to control the gpio pins on the pi with Alexa.
Thanks in advance.

emnik, 2018-03-04:
The node.js example is sample code to link node.js with sinric as we do
with arduino, wemos etc. To control the gpio pins you have to implement
your own node.js application that would control a relay or something. To
access the gpio pins you need a node.js module like  pi-gpio: (
https://github.com/rakeshpai/pi-gpio). There are others too so google it!

malvar0929, 2018-03-04:
Thank you all for the replies. This should enough to get me started. I really love how well it works with Arduino.

closed 2018-03-06 by kakopappa