kakopappa/sinric #39: Report back thermostat's data to Alexa - This is essential...

emnik, 2018-03-04:
Hi! we need to be able to report the changing temperature from the thermostat back to alexa, as this is the only way to have alexa's temperarure value correnspond to the real one. 
**This will be really useful as:**
We would be able to ask alexa what's the temperature and get a responce
We would be able to ask to make <device> warmer (+1) or cooler (-1)  and get a responce with the new temperature. 

The above 2 functionallities are essential to the remote operation of a thermostat as we want to operate it without looking at it's screen.
So we have to know the current temperature and also have reported back the value that is set.

Also if I ask alexa to set the temperature to 23 and afterwards I manually change the setpoint to 25 then I will get false answers to the above questions as alexa won't know the new setpoint.

kakopappa, 2018-03-04:
hello @emnik 

Thank you for your input.

Right now when you set the temp I store it in the database and when you make it warmer and cooler I change the stored value according to the change and respond to Alexa when it requests.

1. Right now Sinric pushes data to clients and clients cannot push data to Sinric. I am working on this

2. Once you send the temp back to Sinric, I have to send a ChangeReport to Alexa. This is a bit complicated. I will look into this.
 https://developer.amazon.com/docs/device-apis/alexa-interface.html#changereport

kakopappa, 2018-03-04:
How do you communicate with the thermostat and know the current temp when you change it manually ?

emnik, 2018-03-04:
I have build a raspberry pi smart thermostat that has a temperature sensor
connected on its gpio pins (an rtc and a relay) and it also gets
temperature data from remote sensors xbees via the xbee protocol and esps
via mqtt. All these are connected via a node.js/express app that runs on
the pi.

kakopappa, 2018-03-06:
@emnik @MacSass

I have updated the server to support changes from ESP devices. Take a look at the example code here

https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example.ino
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino

Let's say you used a push button to turn on/off the switch manually.  When you change the status, you have to call the setPowerStateOnServer function with device id and status, This will update the status on the server.

When Alexa request status of the switch I will use this status to report back. 

Let me know how it goes.

BoriKing, 2018-03-06:
Thanks @kakopappa 

Just what i need it.

MacSass, 2018-03-06:
Hi @kakopappa,
this in fact sounds great - have to play with it to see how it works. Any limitation how often the server can be updated? Is once every 2 minutes OK? Or too often?

Just a small comment - I think there might be a small typo in the example code:

```cpp
//eg: setPowerStateOnServer("deviceid", "CELSIUS", "25.0")
void setTargetTemperatureOnServer(String deviceId, String value, String scale) {
```

should that not be:
```cpp
//eg: setPowerStateOnServer("deviceid", "25.0", "CELSIUS")
void setTargetTemperatureOnServer(String deviceId, String value, String scale) {
```

Minor thing though ...
I´ll test that when I can ...

Regards - MacSass

kakopappa, 2018-03-07:
@MacSass Thanks. I will get this fixed. 

ideally I think it should change it whenever power state or temperature changes. Fewer calls to the server is better as you know no of connected devices to the server keep going up.

emnik, 2018-10-27:
Hi Kakopappa!
After a while I found some time to get back to my thermostat - alexa integration ;-)
I' ve already implemented what you have at your example but you only provide a way to change the TARGET temperature.

We also need a way to update the server for the CURRENT temperature.

As it is right now, it seems that it uses the target temperature value also for the current temperture and this does not make sense...

Also, is it possible the temperature change step to be 0.1? (Now is 0.5)?

Thanks in advance!

MacSass, 2018-11-16:
Hi @kakopappa,
getting back to work on my ESPs finally and tend to agree with @emnik that it would be nice if the current temperature could be set.
To be honest I don´t see if / how this would be possible - could you give some hints how that could work?

Thanks in advance - MacSass

vaxus1, 2018-12-17:
hi. I already tried sinric Thermostat integration with ESP - works pretty well.

I aim to implement weekly scheduling on ESP, but I checked and sinric is not implemented entire "thermostat" propocol API. Is that right?
There are more parameters for scheduling that described in amazon 
https://developer.amazon.com/docs/device-apis/alexa-thermostatcontroller.html

kakopappa, 2018-12-18:
@vaxus1 Yes scheduling is not implemented now. 

I just looked at the scheduling and it requires to send  "supportsScheduling": true if scheduling is supported. by the device. I will add this feature to future iteration.

kakopappa, 2019-01-05:
1. supportsScheduling has been added.  When I respond back to Alexa, supportsScheduling is set to true by default. You might have to delete the device and recreate it if it does not work on the first time. (happened to me)

User: Alexa, set the kitchen AC to 25 degrees for 4 hours.

You will see something like 

[WSc] get text: {"deviceId":"xx","action":"SetTargetTemperature","value":{"targetSetpoint":{"value":25,"scale":"CELSIUS"},"schedule":{"duration":"PT4H"}}}

in the console

2. Temperature and humidity reporting added.
check the example https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example_with_dht_11.ino

MacSass, 2019-01-05:
Nice - I´ll try that temperature and humidity reporting soon ...

Deimos1994, 2019-01-05:
@kakopappa humidity now works with my Google Home, thanks!

By the way for those who use thermostat just like a sensor, set both `setPoint` and `ambientTemperature` with your temperature sensor readings, then it will just report temperature like **It's currently -4.5 degrees.**. But if `setPoint` and `ambientTemperature` have different values then it will say somethng like: **Cooling is set to 25, with a current temperature of -4.5**. (it is valid for google home, don't know about alexa).

@emnik 1. I think you will not recieve any json when It reaches the target. With Alexa you can just set  target temp and ask what current temp is. Alexa is not a thermostart, Alexa is a voice control for a thermostart.

emnik, 2019-01-05:
@Deimos1994 
thanks for your answer! I know that Alexa is just a voice control for a thermostart but I thought I'd ask :-) 
If you think about it it has ALL the data to operate like one and the "real" thermostat would then just have to implement the ON/OFF state with a relay! But OK I'm good as it is ;-)

Deimos1994, 2019-01-05:
@emnik, yes, but what would happen if you lose internet connection for let's say 10 hours?... if the last command from Alexa was "Turn ON" then it will be ON as long as you don't have the internet connection.

emnik, 2019-01-05:
@Deimos1994 Then you would be feeling very very warm ;-) and your wallet would become much thinner :-)

Deimos1994, 2019-01-13:
@racesking the sketch just give you ideas about functionality that sinric can give you, it is not complete.
I think the first step would to build the thermostat (without Sinric)... you need to write code that reads the temperature from a sensor and compares that temperature with the target temperature and decides when to turn the relay on or off.
After you make your thermostat working (and you've fixed all bugs) you can then add Sinric.

It all is not difficult, but if you new to all of this, you cannot learn it all in one day, it takes time.

About Fahrenheit & Celsius
in the sketch we can see:
`//eg: setSetTemperatureSettingOnServer("deviceid", 25.0, "CELSIUS" or "FAHRENHEIT", 23.0, 45.3)`
So it means when you call `setSetTemperatureSettingOnServer()` one of the argument should be a string "CELSIUS" or "FAHRENHEIT" that is how you choose. 
Also it says:  `// Call ONLY If status changed. DO NOT CALL THIS IN loop() and overload the server.` so you shoud know what that means before using Sinric.

I don't know about your level: but before you start you should definetly learn what function is, what variable is and their types. 
I also new to all of this I guess, it has been more than one year since I started to write code, my sketch is 4000 lines long lol, but I still not sure if I want to learn c++

kakopappa, 2019-01-13:
@racesking  There are a lot of how to tutorials on YouTube. Go through them and learn to turn on/off the relay. Thereafter you can integrate with Sinric using the switch template.

Rumpl-X, 2020-01-19:
Hello together,

it will not work, i have put the function so in the loop that it only runs when the tempratur is changing::

```
  if (NmapValue != mapValue) {
        Serial.println("TEMP");

    i = mapValue;
        int temp,integer,count=0,x,cnd=0;
    char ascii[10]={0};
     if(mapValue>>31)
     {
     /*CONVERTING 2's complement value to normal value*/    
     mapValue=~mapValue+1;    
     for(temp=mapValue;temp!=0;temp/=10,count++);    
     ascii[0]=0x2D;
     count++;
     cnd=1;
     }
     else
     for(temp=mapValue;temp!=0;temp/=10,count++);    
     for(x=count-1,temp=mapValue;x>=cnd;x--)
     {

        ascii[x]=(temp%10)+0x30;
        temp/=10;
     }
    Serial.println(ascii);
setTargetTemperatureOnServer("5e2466c60c04793a3a801b59", ascii)
    setPowerStateOnServer("5e2466c60c04793a3a801b59", ascii);
  }
  Temperatur_Zaehler = i;
  Temp = i;
  NmapValue = mapValue;
```
```
all works fine (the serial monitor prints the action, alexa tells me waiting about sinric) but the temprature dont changing.

please help me!!!

greeting

johannes