kakopappa/sinric #40: how to use data from "SetColor"?

UK-Jezz, 2018-03-04:
Hi there, does any one have a small example or can explain how u use the returned data from "SetColor". I have looked at the examples but cant figure it out sorry. I have never used the ArduinoJson Library before.
This is what i currently have in my code, but the serial monitor does not print out the hue value.

        else if(action == "SetColor") {
          String hue = json ["value"]["hue"]["value"];
          Serial.println (" Color is set too: " + hue);          
        }
I am trying to get the hue value so i can send it to a strip of WS2812b LED's using the FastLED library.
Thanks in advance...

kakopappa, 2018-03-04:
Here you go

```
else if(action == "SetColor") {
            // Alexa, set the device name to red
            // get text: {"deviceId":"5a87bd68d82c73628f60cefc","action":"SetColor","value":{"hue":0,"saturation":1,"brightness":1}}
            String hue = json ["value"]["hue"];
            String saturation = json ["value"]["saturation"];
            String brightness = json ["value"]["brightness"];

            Serial.println("[WSc] hue: " + hue);
            Serial.println("[WSc] saturation: " + saturation);
            Serial.println("[WSc] brightness: " + brightness);
        }
```

UK-Jezz, 2018-03-04:
Thank you Kakopappa. I can see i was way off the mark.

nivagc53, 2018-03-04:
Hello Kakopappa,
I'm having a similar problem as UK-Jezz
I want to set the brightness of the WS2812B leds.
Having used the code below
[code]

        else if(action == "setBrightness") {
         String brightNessValue = json ["value"]["?"];
         if (brightNessValue =="?") { 
         setBrightnessValue(deviceId);}
        }

[/code]

Where I have placed a ? do I need to leave this blank?
At the top of the code i have inserted

[code]

      int brightNessValue; // makes a empty "bucket" for brightness 

[/code]

and for the led code 

[code]

      void setBrightnessValue(String deviceId) {
      if (deviceId == "5a9a5576b950f610a7e66214") // Device ID of first device
      {  
      Serial.print("Turn on device id: ");
      Serial.println(deviceId);
      Serial.print("Brightness Valve is ");
      Serial.println(brightNessValue);
      strip.setBrightness(brightNessValue);}
      strip.show();
      }

[/code]

For some reason when I send a "brightness set to ..." from Alexa/sinric the value I receive is the same in the serial output.
However the int valve stays at 0
I've been working on this all day with not a lot of luck. I'm I barking up the wrong tree?
How do I get the "value" of the setBrightness sent from Alexa/sinric into a int?

UK-Jezz, 2018-03-04:
@nivagc53 

Try changing your code from
```
    else if(action == "setBrightness") {
     String brightNessValue = json ["value"]["?"];
     if (brightNessValue =="?") { 
     setBrightnessValue(deviceId);}
    }
```
to
```
    else if(action == "setBrightness") {
     String brightNess = json ["value"]["setBrightness"];
     brightNessValue = brightNess.toInt(); // Convert the String "brightNess" to an integer
    }
```
I have not tested this yet, but it may be what you are looking for.

nivagc53, 2018-03-09:
Hey Jazz
I've tried the above code in several different configurations but still no luck with storing the brightness value in the integer

kakopappa, 2018-03-10:
@UK-Jezz 

Not sure i am late. This should work

    ```
else if(action == "SetBrightness") {
            String brightness = json ["value"];
            Serial.println("[WSc] brightness: " + brightness);
        }
        else if(action == "AdjustBrightness") {
            String brightnessDelta = json ["value"];
            Serial.println("[WSc] brightnessDelta: " + brightnessDelta);
        }
```
output

[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
[WSc] get text: {"deviceId":"5a87bd68d82c73628f60cefc","action":"SetBrightness","value":100}
[WSc] brightness: 100
[WSc] get text: {"deviceId":"5a87bd68d82c73628f60cefc","action":"AdjustBrightness","value":-25}
[WSc] brightnessDelta: -25

nivagc53, 2018-03-15:
Thanks kakoppa 
I've now managed to be able to set the brightness on my WS2812B's!
Working on colour now but looks like BoriKing has helped me out with that one!
Big thanks BoriKing

strider460, 2018-05-21:
i was working on something similar and it took a bit of fiddling around but to get the SetColor command from HSB to RBG. I converted the output to floating point variables and ran them through a function from a color converter library (https://github.com/julioterra/HSB_Color/blob/master/HSBColor.cpp). 

converting the setColor variables to something usable :
```
`     else if(action == "SetColor") {
            // Alexa, set the device name to red
            // get text: {"deviceId":"xxxx","action":"SetColor","value":{"hue":0,"saturation":1,"brightness":1}}
            String hue = json ["value"]["hue"];
            String saturation = json ["value"]["saturation"];
            String brightness = json ["value"]["brightness"];
H=hue.toInt();
S=saturation.toFloat()*100.0;
B=brightness.toFloat()*100.0;
HSBtoRGB(H,S,B,rgb);`

```

the function to take in HSB and output an array (rgb[3]) in terms of 0-255:
```
`void HSBtoRGB(int hue, int sat, int bright, byte rgb[]) {
    
  // constrain all input variables to expected range
    hue = constrain(hue, 0, 360);
    sat = constrain(sat, 0, 100);
    bright = constrain(bright, 0, 100);

  // define maximum value for RGB array elements
  float max_rgb_val = 255;

  // convert saturation and brightness value to decimals and init r, g, b variables
    float sat_f = float(sat) / 100.0;
    float bright_f = float(bright) / 100.0;        
    int r, g, b;
    
    // If brightness is 0 then color is black (achromatic)
    // therefore, R, G and B values will all equal to 0
    if (bright <= 0) {      
        rgb[0] = 0;
        rgb[1] = 0;
        rgb[2] = 0;
    } 
  
  // If saturation is 0 then color is gray (achromatic)
    // therefore, R, G and B values will all equal the current brightness
    if (sat <= 0) {      
        rgb[0] = bright_f * max_rgb_val;
        rgb[1] = bright_f * max_rgb_val;
        rgb[2] = bright_f * max_rgb_val;
    } 
    
    // if saturation and brightness are greater than 0 then calculate 
  // R, G and B values based on the current hue and brightness
    else {
        
        if (hue >= 0 && hue < 120) {
      float hue_primary = 1.0 - (float(hue) / 120.0);
      float hue_secondary = float(hue) / 120.0;
      float sat_primary = (1.0 - hue_primary) * (1.0 - sat_f);
      float sat_secondary = (1.0 - hue_secondary) * (1.0 - sat_f);
      float sat_tertiary = 1.0 - sat_f;
      r = (bright_f * max_rgb_val) * (hue_primary + sat_primary);
      g = (bright_f * max_rgb_val) * (hue_secondary + sat_secondary);
      b = (bright_f * max_rgb_val) * sat_tertiary;  
        }

        else if (hue >= 120 && hue < 240) {
      float hue_primary = 1.0 - ((float(hue)-120.0) / 120.0);
      float hue_secondary = (float(hue)-120.0) / 120.0;
      float sat_primary = (1.0 - hue_primary) * (1.0 - sat_f);
      float sat_secondary = (1.0 - hue_secondary) * (1.0 - sat_f);
      float sat_tertiary = 1.0 - sat_f;
      r = (bright_f * max_rgb_val) * sat_tertiary;  
      g = (bright_f * max_rgb_val) * (hue_primary + sat_primary);
      b = (bright_f * max_rgb_val) * (hue_secondary + sat_secondary);
        }

        else if (hue >= 240 && hue <= 360) {
      float hue_primary = 1.0 - ((float(hue)-240.0) / 120.0);
      float hue_secondary = (float(hue)-240.0) / 120.0;
      float sat_primary = (1.0 - hue_primary) * (1.0 - sat_f);
      float sat_secondary = (1.0 - hue_secondary) * (1.0 - sat_f);
      float sat_tertiary = 1.0 - sat_f;
      r = (bright_f * max_rgb_val) * (hue_secondary + sat_secondary);
      g = (bright_f * max_rgb_val) * sat_tertiary;  
      b = (bright_f * max_rgb_val) * (hue_primary + sat_primary);
        }      
        rgb[0]=r;
        rgb[1]=g;
        rgb[2]=b;
    
    }
}
`
```

Hopefully this points someone in the right direction if they're stuck!