kakopappa/sinric #41: Custom color layout

BoriKing, 2018-03-06:
@kakopappa 

Is it possible you could add a custom color layout to: light devices, set color section?
it  doesn't matter how you set the ("hue": , "saturation": , "brightness":) ill be able to work around it. i just don't want to use the pre-define colors for my custom color, as i'm using them for those specific colors.

kakopappa, 2018-03-07:
Alexa API is not clear about setting custom colors. 
https://developer.amazon.com/docs/device-apis/alexa-colorcontroller.html

Can you do this using app ?

BoriKing, 2018-03-07:
On my side i'm using the values of ("hue": , "saturation": , "brightness":) to call the colors of my choice . Even if the value is, example:  Black ("hue": 0 , "saturation": 0 , "brightness": 0) i could always use those values to call a rainbow color. 

Don't know how is it on your end. Thinking maybe you could rename "Black" for "Custom" and use the ("hue": 0 , "saturation": 0 , "brightness": 0) values.

If not possible don't worry about it i could always sacrifice one of the pre-define colors for my animation.

kakopappa, 2018-03-07:
i dont think this is possible with the current apis. I will update you if something comes up

closed 2018-03-07 by kakopappa