kakopappa/sinric #43: "Not an issue" but a video example

BoriKing, 2018-03-08:
Sorry for the low quality video i'm using.

Here is a video example to control an rbg ring using sinric.(**code not included**)
1st is a rainbow color then the pre-define colors from the alexa app
[RBG ring](https://www.youtube.com/watch?v=Zz7hz9FOjHQ)

nivagc53, 2018-03-09:
Wow. Fantastic 
Could you post the code you used?

BoriKing, 2018-03-15:
@nivagc53 
Little late but...Here you go.
[RGB led's](https://github.com/BoriKing/Sinric-WiFi-Manager/blob/master/Arduino%20ESP-8266%20examples/RGB-LEDs/Sinric_Multi_WiFi_Networks_and_CredentialsReset.ino)

nivagc53, 2018-03-15:
Thank you so much
I manage to get the "setBrigthNess" working 2 weeks ago and have been working on the colour control since!
Fantastic work

UK-Jezz, 2018-03-22:
@BoriKing 
Thanks for sharing your code. I was interested in seeing how you produced the colours.
I have the same result, but using a different method.

        else if (action == "SetColor") {
          String hue = json ["value"]["hue"];
          String saturation = json ["value"]["saturation"];
          String brightness = json ["value"]["brightness"];

          // Convert string "saturation" to a floating point then map it in the range of 0 to 255
          float satCon = saturation.toFloat();
          satCon = satCon * 10000; // Move the decimal 4 places to the Right
          satCon = map(satCon, 0, 10000, 0, 255);
          // Convert string "brightness" to a floating point then map it in the range of 0 to 255
          float briCon = brightness.toFloat();
          briCon = briCon * 10000; // Move the decimal 4 places to the Right
          briCon = map(briCon, 0, 10000, 0, 255);
          // convert string "hue" to an integer between 0 and 255
          int hueCon = hue.toInt();
          hueCon = map(hueCon, 0, 359, 0, 255);

          // Assign values to the LEDs and then illuminate them
          CHSV hsv(hueCon, satCon, briCon);
          CRGB rgb;
          hsv2rgb_spectrum(hsv, rgb);
          for (int led = 0; led < (NUM_LEDS / 2); led++) {
            leds[led] = rgb;                  // use rgb for FastLED's Spectrum colour map or
            leds[(NUM_LEDS - 1) - led] = rgb; // use hsv for FastLED's Rainbow colour map
            FastLED.show();
            FastLED.delay(1); // create a wipe effect
          }
        }

BoriKing, 2018-03-22:
@UK-Jezz 

Thanks for sharing. Very well implemented.  However some of the colors on my rgb ring/ strips look the same eg: (lavander and purple) so i decided to call each color individually using the rgb values of the colors i really wanted in place.
Eg: Purple rgb value = 128,0,128 wich is the one alexa uses as hsb value, but instead I'm using the Indigo rgb value= 75,0,130 for my ring/strips as it will have the closest purple look on my ring/strips.

Again, thanks alot.

closed 2018-03-29 by BoriKing