kakopappa/sinric #44: Alexa.PlaybackController

BoriKing, 2018-03-10:
@kakopappa 

I'm working on an Alexa IR controller for TV's, so far the following things are working:

- setPowerState
- SkipChannels
- SetMute
- AdjustVolume
- SetVolume
- Stop
- Play
- Pause
- SelectInput
- ChangeChannel

However the following things aren't working: (at least for me)

- StartOver
- Previous
- Next
- Rewind
- FastForward

https://developer.amazon.com/docs/device-apis/alexa-playbackcontroller.html

kakopappa, 2018-03-10:
> playbackcontroller.html

You are correct, these are not supported. I have not added these in the
response back to alexa (when discovering the device). do you need these?

BoriKing, 2018-03-10:
Whenever you can, i'm working on multiples projects atm so no rush.

closed 2018-03-29 by BoriKing