kakopappa/sinric #45: Door Opener

Forbescraft, 2018-03-10:
Any plan to set up something that responds to "open" and "close" commands?

kakopappa, 2018-03-10:
Sure. I assume you want the LockController.

I will update this issue again when it's done

Forbescraft, 2018-03-12:
Wow! Thanks for the quick response. I was actually wanting to do a garage door opener. Looking through the capability interface reference they may not have implemented what I need yet, which would be open/close as apposed to on/off or lock/unlock.

BrainRift, 2018-03-16:
I am looking for something similar.  Right now I was testing it with the "switch".  I added a delay after the "on" to then immediately activate the "off"  delay(500) seems to work fine.  That makes the relay act as a temporary button, but the voice commands and the display from Alexa app using the Sinric skill only display as on or off.

PStewardYUL, 2018-03-17:
I swapped from the Wemo version to Sinric today and it's working great! I tried the lock function and it definitely does work but appears that it's missing that response part (as Alexa says it's not ressponding even though it locks!).
It would be great to have more control over trigger words but I'm not sure it's possible with Alexa yet...

dhartis78, 2018-07-31:
Hello is there an update to this.?

Kittyjuls, 2018-10-09:
I would be interested too, I am able to open and close my garage door using a wemos wired to a remote but having to say 'Alexa, turn on the garage door' is just taking the shine off of the skill. All I want is to replace on / off with 'open / close'.

Forbescraft, 2018-10-09:
Precisely.

computerfreek274, 2019-01-06:
Any code for a relay momentary switch for say a door just need it to click the relay.