kakopappa/sinric #49: ESP-01?

paulrjacques, 2018-03-15:
Hi,
   I installed the switch_example software on an ESP-01, I see Alexas commands on the Serial monitor. How do I control an IO output pin? I looked at both outputs (IO00 and IO02). Am I supposed to write the software to drive one of these pins?

nivagc53, 2018-03-15:
Hi paulrjacques,

Sound like your starting out like me a few months ago?
Take a look at this YouTube video. Helped me no end!
https://youtu.be/ZyetXihGrM0
Hope this helps

paulrjacques, 2018-03-15:
Hi kakopappa,

   Ah-Ha! All that sinric sample code is very generic, I just need to add
the code to control the outputs! I can't wait to get home and try it!

   Sinric seems like the way to go. Like you and many others, I too felt
the frustration of trying to get your last ESP project working using
the Echo Version 2. Too many variations to consider.

My girlfriend planted a variety of fruit trees and we just can't get into
a routine to remember to go outside and turn the drip system on and off so
I want to aid the effort. My next project will be to add ground moisture
sensors to automate it and somehow have Alexa remind us about the on-off
state of the drips.

Keep up the good work!

Thank You!!
Paul Jacques

> Hi paulrjacques,
>
> Sound like your starting out like me a few months ago?
> Take a look at this YouTube video. Helped me no end!
> https://youtu.be/ZyetXihGrM0
> Hope this helps
>
>
> https://github.com/kakopappa/sinric/issues/49#issuecomment-373268648

kakopappa, 2018-03-17:
@nivagc53 Thanks for sharing the video. It was nice to see Sinric featured in this video. one day people will write songs about Sinric! 

@paulrjacques  Thank you.

Sinric gives you butter, sugar, and flour. You have to make the cake! haha I tried to make it generic as much as possible to it can be used on wide range of devices.

closed 2018-03-19 by kakopappa