kakopappa/sinric #50: sinric NodeMCU

dti1, 2018-03-17:
Hi 

I'm Having issues getting this to run with the NodeMUC that i use.

a small back story i had this working with 2 relays using the Wemos code that you did a few months back and it was flawless for quite a time. then all of a sudden my relays kept switching them selfs off with no command from anyone in the house and as they was on the TV it was not really working out. I ended up moving away from that.

As i have been off work i though i would revisit setting the relays back up and have come accross Sinric. I have signed up to the site got an API key and a set of Device Id's for the relays. I upload the sketch and it seems to go onto the NodeMUC fine. I open the serial monitor and can see that i'm reaching out to sinric server. I even run a test from sinric back to the Node MCU and can see the test happening in real time.

Problem starts here - i can not get Alexa discovery to find the relays. 
I'm thinking either this wont work with the NodeMCU or more likely its something i'm doing wrong. any help would be greatly appreciated

kakopappa, 2018-03-17:
Did you install the skill?

dti1, 2018-03-17:
yes i have the skill and i can see it on my Alexa App

kakopappa, 2018-03-17:
If you followed the instructions on home page it should work for you. I
tested it couple of hours ago after I pushed the Google Home update and all
seems fine.

1. Make sure skill is installed and linked to your account.

If still does not work,

2. Try creating a new sinric account (email address doesn't have to be
real. Sinric does not validate)

Create a new smart home device in sinric

Unlink the existing skill and link again with new account

Change the API key in the sketch and flash again.

Try discovering

I am outside now, I can take a look at the logs when I get back

dti1, 2018-03-17:
Worked a treat. i can now see devices with Alexa. i need to set the pins up on my board now but will have to look at that later. thanks loads for the help. will make a visit to your paypal later when i get back

once again thanks for the help 👍

dti1, 2018-03-18:
I've come back to this today and i seem to be stuck again. short version i can now see everything working from a server point of view. serial monitor is saying everything i need it to day E.g "TV on" on command and TV off on command.

i've defined a relay pin below serial print

#define  Relay D1

Serial.print("TV on: ");
Serial.println(deviceId); 
digitalWrite(Relay, HIGH); 

i've set the node up to D1 with power and i can not get the relay to trigger but the relay board has power.

i'm guessing i'm stuck at the pin selection process.

if you can help me with this it would be great.

kakopappa, 2018-03-18:
What's the board model?

Make sure relay is not connected main power

Make sure it's wired proper: e.g. NO , NC

Try setting it to low instead of high

BoriKing, 2018-03-18:
Have you call the relay on setup function?
E.g:

void setup() {
  Serial.begin(115200);
  Serial.println();
  pinMode(Relay, OUTPUT);
  delay(10);
  digitalWrite(Relay, LOW);

dti1, 2018-03-18:
Guys You've both nailed it for me

I changed the Low to High and i didnt have the OUTPUT in worked 1st time

Great help lads 🥇

closed 2018-03-19 by kakopappa