kakopappa/sinric #51: discovery failed.

hamm3rs, 2018-03-19:
I am not able to add device to alexa.
Im getting this on the serial monitor:

WiFi connected. IP address: 192.168.1.70
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
[WSc] get text: {"action":"test","sender":"web"}
[WSc] received test command from sinric.com

the test is from me using the test button on the sinric.com website.
I have the sinric skill added to alexa.

any help would be appreciated.

BoriKing, 2018-03-19:
Does your device id from sinric/sketch match? If so unlink skill, re-link skill using the same exact email address and password you used in sinric.com.

hamm3rs, 2018-03-19:
I have tried re linking the skill as well.  link says successful.

kakopappa, 2018-03-20:
stragne. normally works fine.

1. try creating a new account.

2. drop me your user/pass. I will take a look when i get back home (aruna.tennakoon@gmail.com)

hamm3rs, 2018-03-20:
I think its cause i don't have an alexa devise, i just thought i could use my phone to send voice commands.

BoriKing, 2018-03-20:
The Alexa app doesn't work independently you need an Amazon alexa/echo device.

nilavachowdhury, 2018-03-21:
@hamm3rs 

1) Visit https://echosim.io/ 
2) Login with your amazon account 
3) Login with that same account in alexa app in phone
4) Link sinric skill and discover devices 
5) Test out by sending voice command from alexa app!

hamm3rs, 2018-03-21:
That worked out just great.  Thank you so much.

gouthamraj91, 2018-04-29:
Worked for me too, thanks!

arshad3699, 2018-05-31:
can any body share code working one