kakopappa/sinric #52: Device / Light State (Turn on/turn off)

JanB97, 2018-03-25:
Hi,
I was wondering why my Wemos does not work anymore properly.
For some reason if I use the window shutters option in sinric.com to create a device, I do not get a powerstate anymore, it just sends a percentage to my wemos.
Is that something you changed or is that a problem with amazon?

kakopappa, 2018-03-25:
Hi,

I didn't change the code for about a week now. How do you handle it?

For shutters I have added

   - Window Shutters

   1. Alexa.PowerController
   2. Alexa.ChannelController
   3. Alexa.PercentageController

If you say turn on/off [devicename] it should send On or Off

JanB97, 2018-03-25:
Ok, i guess Amazon changed the way they handle these then.

I wrote my code so I can change the ID of each device at the top of the script without changing it everywhere I use it and I'm glad i wrote it this way now :)

I had some routines set up for the shutters, as the translation was not really good and I wanted to make my own commands. 
I just created new devices as switches in sinric and changed the IDs in the code.

If I set a routine with the command "turn on, 100%" I always also got a turn on event in code, but now it only sends the command for setting the percentage 100%. 
As I can not get rid of the percentage with these devices I had to change it to switch now.

I guess I could have handled the percentages as turn on event, but that would have been more work at the end.