kakopappa/sinric #53: Webhook reconnect issues after 2.1.0 update

nilavachowdhury, 2018-03-27:
Webhook doesn't reconnect after disconnection after updating arduinowebsocket library to v 2.1.0

kakopappa, 2018-03-28:
sounds like arduinowebsocket  issue ?

nilavachowdhury, 2018-03-28:
Reverted back to old library, still same issue , it reconnects only when WiFi connection drops , doesn't reconnect when web socket disconnects due to no internet.

nilavachowdhury, 2018-03-28:
Seems like an issue with the library , however found a solution from the original creator of the library .
He mentioned "a connection loss can only be detected if you send data ( this is a limitation of the TCP stack on the ESP).
you can use webSocket.sendPing(); to check the if the connection is alive.
if you call it for example every 10 sec you can detect the loss of the connection fast.

reducing your HEARTBEAT_INTERVAL will do the same trick but with a little more data rate."

I created a timer to call for the ping event and it works great now!

closed 2018-04-03 by nilavachowdhury