kakopappa/sinric #55: wemos board control with Alexa

4x1ks, 2018-04-01:
Newbie alert...I have an arduino d1 mini.  I added the sinric skill to alexa smart home.  I uploaded code to turn the BUILTIN_LED on and off.  The mini shows up on my wifi.  The alexa app finds my device "bigrelay."

When I ask Alexa to turn on or off the device, she says "ok"  but nothing happens.  The LED does not change state.

I think my code is ok, it isn't mine..I copied it.  I also tried the code to control 5v to a pin to a relay and this also did not work.  I'm thinking that it isn't the code, it is the Alexa / Wemos connection.

Any help is appreciated very much.

Mark

kakopappa, 2018-04-01:
When you ask to turn on or off does it print anything in serial console?

4x1ks, 2018-04-01:
> Thanks for writing me.
>
> This is what get when I upload.  When I ask to turn on or off the device I
> get no response on the serial print.  But the Alexa app does show the
> device on or off...and when I upload the sketch I get a momentary click of
> the relay that is attached to the wemos...after the sketch is fully
> uploaded there is no change in the relay or lights at all....

Thanks for writing me.

No text in serial print except for the WIFI connecting message and my Lan
port..  When I ask to turn on or off the device I get no response on the
serial print.  But the Alexa app does show the device on or off...and when
I upload the sketch I get a momentary click of the relay that is attached
to the wemos...after the sketch is fully uploaded there is no change in the
relay or lights at all....

4x1ks, 2018-04-01:
this is my code

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include "fauxmoESP.h"

#define WIFI_SSID "Sura Office"
#define WIFI_PASS "password"
#define SERIAL_BAUDRATE                 115200

fauxmoESP fauxmo;

//
-----------------------------------------------------------------------------
// Wifi
//
-----------------------------------------------------------------------------

void wifiSetup() {

    // Set WIFI module to STA mode
    WiFi.mode(WIFI_STA);

    // Connect
    Serial.printf("[WIFI] Connecting to %s ", WIFI_SSID);
    WiFi.begin(WIFI_SSID, WIFI_PASS);

    // Wait
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(100);
    }
    Serial.println();

    // Connected!
    Serial.printf("[WIFI] STATION Mode, SSID: %s, IP address: %s\n",
WiFi.SSID().c_str(), WiFi.localIP().toString().c_str());
}

void callback(uint8_t device_id, const char * device_name, bool state) {
  Serial.print("Device "); Serial.print(device_name);
  Serial.print(" state: ");
  if (state) {
    Serial.println("ON");
    digitalWrite(D4,HIGH);
  } else {
    Serial.println("OFF");
    digitalWrite(D4,LOW);
  }
}

void setup() {
    // Init serial port and clean garbage
    Serial.begin(SERIAL_BAUDRATE);
    Serial.println("FauxMo demo sketch");
    Serial.println("After connection, ask Alexa/Echo to 'turn <devicename>
on' or 'off'");

    pinMode (D4,OUTPUT);

    // Wifi
    wifiSetup();

    // Fauxmo
    fauxmo.addDevice("Fish Tank");
    fauxmo.onMessage(callback);
}

void loop() {
  fauxmo.handle();
}

kakopappa, 2018-04-01:
You are using fauxmoESP library.. this is nothing to do with sinric.

If you want to use Sinric:

Go to sinric examples in the github repo

Copy the switch example

Change the API key

Flash

Ask Alexa to discover devices

kakopappa, 2018-04-01:
Follow this youtube link to install Arduinojson. 
https://www.youtube.com/watch?v=GUTpaY1YaXo

do the same for websocket

it's not that complicated.

4x1ks, 2018-04-01:
I haven't programmed since college....35 years...thanks...I will try

4x1ks, 2018-04-01:
websocketsclient.h     is not a zip file to be uploaded into the library.
It is a page of code.  So how do you get it into your library?  The
arduino.json was easy because it was already in my library.

thanks

4x1ks, 2018-04-01:
I managed to put websocketsclient.h and the json onto the sketch.  It
compiled with no errors.

Where can I give a name to the switch?  For example, if I want to call it
"relay" or "wemosrelay"

In the other code, I was able to give it a name.

thanks

4x1ks, 2018-04-01:
in the serial print it started printing

[MAIN] Free heap: 45088 bytes

over and over and over...continuously....but at least it is talking to the
serial print

kakopappa, 2018-04-01:
Goto sinric.com

Create an account

Create a device inside Sinric. There you can give a device name

Copy the API key in the dashboard.

Paste the API key in the arduino sketch. Flash

Install the Srinric smart home skill using Alexa app

Link the skill using above created account

Then you can use turn on / off device name commands

Step by step instructions also on sinric.com home page

4x1ks, 2018-04-01:
> AkNvqClqTHaYcwRczSXcNr8cIs0lAxXUks5tkMjTgaJpZM4TCwfP

ok...you're a genius.  The commands are showing up in the serial print.  on
and off...

but....how do I define the pin that will control the relay?  Is there a
line in the code?  There must be....

4x1ks, 2018-04-01:
> 377783655>,

i think I need the "set power state" code...there is sample code at amazon,
but can I use that?

kakopappa, 2018-04-01:
Learn to turn on / off relay in another sketch.

There are lots of examples already on internet.

Then you can copy the code to sinric that will be easy .

Be careful when you are playing with high voltage

4x1ks, 2018-04-01:
> 377786153
> AkNvqOv9l2GugQwEA6IapoX1evklgJCIks5tkNQzgaJpZM4TCwfP

when you say " Then you can copy the code to sinric that will be easy ."

you mean copy the code into the sketch?  There is no way to copy the code
to sinric ..you must mean to write it into the sketch, yes?

I think I am very close to having it work now.

are you referring to the code that is at:

developer.amazon.com/docs/device-apis/alexa-powercontroller.html    ??

4x1ks, 2018-04-01:
these are the error messages I got from the code I copied from Amazon's
website on power setting.  Do these error codes make sense to you?

_________________________________________________________________________________________________________________

Arduino: 1.8.5 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 4M (1M
SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Users\Radio Room\Desktop\sinric_sketch1\sinric_sketch1.ino: In function
'void webSocketEvent(WStype_t, uint8_t*, size_t)':

sinric_sketch1:81: error: expected ';' before ':' token

   "directive": {

              ^

sinric_sketch1:258: error: expected '}' at end of input

 }

 ^

sinric_sketch1:258: error: expected '}' at end of input

sinric_sketch1:258: error: expected '}' at end of input

sinric_sketch1:258: error: expected '}' at end of input

exit status 1
expected ';' before ':' token

__________________________________________________

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

kakopappa, 2018-04-01:
What I meant was to learn to ten on off relay first and then copy that code
inside sinric sketch

Error seems to be related to a missing ;

4x1ks, 2018-04-02:
good morning

So I went to the site:

// https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html
still not sure how to integrate that code into the existing sketch to make
it work.

I noticed in my sketch that it DOES communicate between alexa and sinric,
by listing the commands on the serial print.

In the github example that I am using, there is a line that says:

// {"deviceId": xxxx, "action": "setPowerState", value: "ON"}

this is a comment line, but

obviously I must fill in xxxx, "action" : setPowerState", value: "ON"
......these variables must be linked somehow to the 4 on / off examples
that are on the
amazon developer site....I can't figure out how to do that.

I also did not find any other on/off sketches for sinric that look like
they will work with the existing sketch.

Any help is greatly appreciated.  I've gotten close to making it work now
that Alexa is identifying voice commands, but obviously that is not enough
to get the d1 mini to trigger the relay.

thanks

Mark

kakopappa, 2018-04-02:
Hello

// {"deviceId": xxxx, "action": "setPowerState", value: "ON"}

xxx here is the device id displayed in your dashboard. device id is used to identify which device is this command related to when you have multiple smart home devices.

4x1ks, 2018-04-02:
hello:

1:  do I need to add the device id number in place of xxxx ...and also
uncomment the line? right now the line has // in front of it and no
numerical device id in place of xxxx

2. I added the following 3 lines...to see if this would solve the problem

#define MYPIN D4

then I also added in the turn device on routine

digitalWrite(MYPIN, HIGH);

and then in the turn device off routine I added

digitalWrite(MYPIN, LOW);

This did not help.  There is still no response form pin d4 on the arduino

kakopappa, 2018-04-02:
1. 

if (deviceId == "your device id from dashboard")
{
Serial.print("Turn off Device ID: ");
Serial.println(deviceId);
}

2. In the setup you have to set the pinmode

void setup(){
   pinMode(MYPIN, OUTPUT);
   .....
 }

4x1ks, 2018-04-02:
>    1. In the setup you have to set the pinmode

It is working!!!!!!  Thanks....I will send you chocolate!

kakopappa, 2018-04-04:
still waiting ..

4x1ks, 2018-04-04:
Yes...it works just great!  Thanks.  If you tell me how to mail it..then I
will send you chocolate!

Next I am going to add more features to the sketch...

kakopappa, 2018-04-05:
becareful when you are working with high voltage.  don't burn your house down

https://paypal.me/arunat

4x1ks, 2018-04-05:
I am a ham radio operator and welder.....I deal with high voltages
often....thanks!

https://paypal.me/arunat

GaryLittlemore, 2018-04-05:
4x1ks I'm also a Noob trying to get this working with a relay shield. Any chance you could post your working .ino code please?

I'm also confused about this device id, the TODO says "MyApiKey", "MySSID" & "MyWifiPassword", it don't mention I have to get a deviceId?

GaryLittlemore, 2018-04-05:
Thanks for that, I've found it. maybe the code needs to have the "// TODO" on it.

Just need the relay code now ;-)

GaryLittlemore, 2018-04-05:
Where in the .ino code should the relay code be placed? Line number?

void setup(){
pinMode(MYPIN, OUTPUT);
// rest of the code below..
}

kakopappa, 2018-04-05:
in setup function you have to set the pinMode. 

then change to high or low to turn on / off

https://arduinobasics.blogspot.com/2014/09/relay-module.html

GaryLittlemore, 2018-04-05:
Thanks I don't really understand that, I'm not a coder. I've been trying to follow along watching videos and replicating. Hopefully 4x1ks will be able to post their code. Don't worry I'm not doing the wiring with high voltage, a friend will be helping me there.

4x1ks, 2018-04-05:
Hey Gary

Send me your email...I will get back to you.

Mark 4x1ks

GaryLittlemore, 2018-04-05:
gary (dot) littlemore (at) btinternet (dot) com

Thanks Mark

nizamaj, 2018-11-05:
Hi Machan kakopappa  , Thanks for the wonderful work ..   i am planing to flash my LinkNode R4 to work with alexa  (https://www.amazon.com/LinkNode-R4-controller-4-Channel-Module/dp/B01FUC9U58/ref=cm_cr_arp_d_product_top?ie=UTF8) and found out your old code .. (https://github.com/CharlesJGantt/LinkNode-R4-Controlled-By-Amazon-Alexa/tree/master) .. My question is .. is there a way i can include sinric to this code ?? or do i need to change entire libraries and codes ?

kakopappa, 2018-11-05:
I think old method still can be used but unrealiable. It used to emulate a Belkin WeMos switch. Sometimes it work sometimes not. 

If you do not have access to Sinric Skill you have to use the old method.

nizamaj, 2018-11-05:
Exactly the old method sometimes it working sometime don't .. planing to flash my LinkNode with new code ..

Dr-Bundolo, 2021-02-01:
First of all Thanks for creating the library and all the time you've spent trying to help all of us newbies.   I've spent the better part of two days trying to get my D1 mini to work with Alexa.  I'm very close.  As you can see from the screen shot of  the serial monitor the sketch works.  If I press test on Sinric, or power on with Alexa it registers on the Com.   Great!  The problem is that it does NOT turn on the relay.  I've tried the relay on numerous pins (changing the pin assignment in the program) with no change in the result.  I've tried numerous boards (right now I'm using WeMos D1 R1, but I've tried the generic board as well.  Still won't trigger the relay.  I've tried different program with the same result.  I've tested the D1 with a blink program - works.  Tried different relays  with the same result.  Any advice
[image]
Here's the Code
`    #include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WebSocketsClient.h> //  https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <ArduinoJson.h> // https://github.com/kakopappa/sinric/wiki/How-to-add-dependency-libraries
#include <StreamString.h>

ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "f0e87a67-bcca-469e-b593-3a2882e60ca3" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "kuggle's_condo_2" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "jakesnak" // TODO: Change to your Wifi network password

#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

const int relayPin = 5;

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "600dce7798845d1692f01e0e") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
   digitalWrite(relayPin, HIGH);
       digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH
  } 
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);  
   digitalWrite(relayPin, LOW);  
  }     
}

void turnOff(String deviceId) {
   if (deviceId == "5600dce7798845d1692f01e0e") // Device ID of first device
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     
     digitalWrite(relayPin, LOW);  // turn off relay with voltage LOW
   }
   else {
     Serial.print("Turn off for unknown device id: ");
     Serial.println(deviceId);    
  }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
[... 131 lines cut ...]
#endif        
  root["action"] = "SetTargetTemperature";
  root["deviceId"] = deviceId;

#if ARDUINOJSON_VERSION_MAJOR == 5
  JsonObject& valueObj = root.createNestedObject("value");
  JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");
#endif
#if ARDUINOJSON_VERSION_MAJOR == 6        
  JsonObject valueObj = root.createNestedObject("value");
  JsonObject targetSetpoint = valueObj.createNestedObject("targetSetpoint");
#endif  
  targetSetpoint["value"] = value;
  targetSetpoint["scale"] = scale;
   
  StreamString databuf;
#if ARDUINOJSON_VERSION_MAJOR == 5
  root.printTo(databuf);
#endif
#if ARDUINOJSON_VERSION_MAJOR == 6        
  serializeJson(root, databuf);
#endif  
  
  webSocket.sendTXT(databuf);
} `

Dr-Bundolo, 2021-02-02:
Okay, A day later I have solved my problem.  As always it was a series of problems.  The first was making sure I had the proper drivers downloaded.  Then it was testing all my USB cables until I found one that worked and the computer recognized the port.  That was a biggie.  But still couldn't get the program to work.  Alexa would turn on and off the device.  You could see the BLUE LED come on when Alexa turned it on, and turn off when Alexa turned it off....  so close.  then after searching more on the internet the final piece fell in place.  As is the usual case I had the relay powered from the VCC.  I checked it with a volt meter and it was getting 4.75.  But the relay would not trigger.  Finally I read someone suggested trying using the 3.3V output on the board.  I was not convinced as my experience is that voltage is too low to trigger the relay, but low and behold it worked!  So If your getting everything but the relay working, change to 3.3v.   It will work.     Hope this helps.