kakopappa/sinric #56: Not Able to login and see my devices

debagchi1978, 2018-04-13:
Hi,
I have an account and some devices configured. But from today morning, I suddenly saw that though sinric.com taking me to my landing page, showing my Username, but no devices are present. 
Then I logged out. I can't login since then with my uid and pwd.It is showing the following error on the screen.

Response with status: 0 for URL: null

Please let me know what has to be done to login and get my configured devices.
Thanks,
Deb

kakopappa, 2018-04-14:
Hi

I just checked the login and I can login. Can you please try again or
create a new login.

Email address doesn't have to be real one. Fake one would do . Make sure to
remember it when linking the skill

debagchi1978, 2018-04-14:
Hi,
I also just checked and found it is working now and its behaving as expected. May be server was down at that time.
But my concern remains, when the server is down, do you have backup server(s) and load balancer which will re-route the requests to the server(s) which is/are up and running? Otherwise, the system will be down and users like me who relies on your solution extensively will be in big trouble.

kakopappa, 2018-04-15:
"when the server is down, do you have backup server(s) and load balancer
which will re-route the requests"

Haha.. No this is a free service. I paid for the server  + Amazon Lambada
services from my own pocket and spent my own time building the Amazon skill
so we all can benefit from it. I try to keep the server up and running all
the time for everyone including my self.

debagchi1978, 2018-04-15:
OK I see. Thanks a lot for your free service to the world. I am closing the issue.

Thanks 
deb

closed 2018-04-15 by debagchi1978