kakopappa/sinric #59: playBackContoller Support

Abulrock, 2018-05-13:
first thank you very much for the great work. This takes alexa to the next level.
i was just wondering if it would be possible to add support on Sinric for PlayBackControl. 
i saw its mentioned here [https://github.com/kakopappa/sinric/wiki/Supported-devices-and-their-capabilities](url) but not sure if its supported already?

thanks

kakopappa, 2018-05-13:
@Abulrock Thanks

Yes. I have added it to Tv and Speaker device types

Abulrock, 2018-05-13:
thanks for the quick reply, that;s great to hear, but when i add a tv device and ask Alexa to Play next on device. Alexa replies saying "TV not support that". 
is there something i should be doing differently?

kakopappa, 2018-05-13:
I just checked the code and database. I am sure it's there

Have you tried creating a speaker type device ?

Abulrock, 2018-05-13:
yes i tried it with the Speaker earlier as well, but Alexa responds with the same thing "not supported"

Abulrock, 2018-05-15:
Any chance this has been fixed? I still can't seem to get PlayBack to work

kakopappa, 2018-05-19:
Hi

I just checked and it seems to work fine. This is what I did

1. Create a new device with type speaker. called it "speaker"
2. Discover for new devices (Alexa)
3. Ask Alexa to Play speaker. 

Supports only,  "Play",  "Pause", "Stop"

Alexa discovery request response:

{
 "endpointId":"xx",
         "friendlyName":"speaker",
         "description":"speaker",
         "manufacturerName":"SINRIC",
         "displayCategories":[  
            "SPEAKER"
         ],
         "cookie":{  
            "deviceId":"xx"
         },
         "capabilities":[  
            {  
               "type":"AlexaInterface",
               "interface":"Alexa",
               "version":"3"
            },
            {  
               "type":"AlexaInterface",
               "interface":"Alexa.PowerController",
               "version":"3",
               "properties":{  
                  "supported":[  
                     {  
                        "name":"powerState"
                     }
                  ],
                  "proactivelyReported":true,
                  "retrievable":true
               }
            },
            {  
               "type":"AlexaInterface",
               "interface":"Alexa.PlaybackController",
               "version":"3",
               "supportedOperations":[  
                  "Play",
                  "Pause",
                  "Stop"
               ],
               "properties":{  
                  "supported":[  
                     {  
                        "name":"input"
                     }
                  ]
               }
},

here is the working demo

https://1drv.ms/v/s!AlN6Rnygw-lfgrVM3YICWZA3a7o1Pg