kakopappa/sinric #61: How do I use The window shutters example?

flappypot, 2018-05-17:
Could you post the code that needs to be added and the GPIO pins that will be used during the setup process? Thanks!

Seaha4487, 2018-07-07:
i also wanna know how to do this

kakopappa, 2018-07-07:
@Seaha4487 GPIOs depend on how you wire your development board. You have to be more specific about your setup

Seaha4487, 2018-07-07:
Its no prblem i can use every gpio. But what have i to change that it works a example would be nice cause i didnt know where to change everything and what to do that alexa knows the status of the blinds

Seaha4487, 2018-07-15:
I absolutly get it not to work please help me. Wich gpios should iuse and what have i to change in the window shutters example that everything works

kakopappa, 2018-07-16:
@Seaha4487 If you want help, you have to tell us 

1. What are you using to open window shutters? 
2. You can use any GPIO pins. If you do not know how to use GOIP pins look at LED example

Here is the code to turn on/off

```
void setup() {
 pinMode(13, OUTPUT); //Declare GPIO13 as output
}
void loop() {
 digitalWrite(13, HIGH);      // turn on 
 delay(1000);                         // Wait 1 second 
 digitalWrite(13, LOW);       // turn off 
 delay(1000);                        // wait 1 second
}
```

Seaha4487, 2018-07-16:
Hey and thx. But i wanna use an stepper motor. I know the gpios and other standard stuff but there are so lot infos in your code

MfG Settari Daniel

  (#61) 

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/kakopappa/sinric/issues/61#issuecomment-405161288",
"url": "https://github.com/kakopappa/sinric/issues/61#issuecomment-405161288",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [kakopappa/sinric] How do I use The window shutters example? (#61)",
"sections": [
{
"text": "",
"activityTitle": "**Aruna Tennakoon**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@kakopappa",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"kakopappa/sinric\",\n\"issueId\": 61,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"kakopappa/sinric\",\n\"issueId\": 61\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/kakopappa/sinric/issues/61#issuecomment-405161288"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 336331360\n}"
}
],
"themeColor": "26292E"
}
]

kakopappa, 2018-07-16:
I think you need how to use sinric guide.  There are lot of examples on youtube
https://www.youtube.com/results?search_query=sinric

Seaha4487, 2018-07-16:
I already watched this but its not for window shutter

MfG Settari Daniel

  (#61) 

https://www.youtube.com/results?search_query=sinric

[
{