kakopappa/sinric #62: Unable to relink my account.

schaffertown, 2018-05-21:
[image]
I have been using  sinric with alexa to control my lights for over a month. This evening they quit responding.  There was a message  of trouble with server. I reset my devices and router. No luck. I unlinked my account and am now unable to relink it. Is the server down?

kakopappa, 2018-05-21:
Hello,

Can you check whether you can access https://alexa.sinric.com from your mobile ? 

I do have a problem where some ips are sending large payloads to the webserver causing the api to crash, so time to time it gets crashed and restarted

schaffertown, 2018-05-21:
I was able to relink an hour after this post. Worked this morning with no problems.