kakopappa/sinric #67: Code example

Jury1985, 2018-06-01:
Hello, i use an  ESP8266 and  i have load the Switch example  for the sinric app.
the voice control over the alexa dot  and the trigger over the handy app is working.
my question is, can some one give me an example how i can read in an hardware switch in the esp8266  that i can switch it on an off. i dont know how imust programm the code .

thanks

AshokBaijal, 2018-06-04:
Hi,
Uploaded a sample code which uses touch switches for controlling the devices in addition to Alexa control. Please see code in my post which raises some other issues

pico563, 2018-07-05:
@AshokBaijal 

> Please see code in my post which raises some other issues

where can i find code?