kakopappa/sinric #68: (1) Update Device status from Sinric server on start up and (2) How to set up a device as a fan

AshokBaijal, 2018-06-04:
I had earlier tried using Kakopappa's Belkin emulator but found the system was not reliable for the echo Dots owned by me. This code has been working robustly. Thanks Kakopappa.

I have used a touch switch so that the device can to switched on or off by touch and can be switched on/off  using Alexa instead of a mechanical switch which retains its state. The touch switch can be replaced with a press switch also. The Sinric server also gets updated. I am attaching the .ino file.

I have two issues for which I need a resolution

1. In India, we have frequent power tripping. The ESP8266 loses its status and switches have to be reset when the power resumes. I want to recover the status of the devices from Sinric server at startup so that the switches can be reset to the state as existed before power failure. Need a function to read the device status just like the function to update the Sinric server when the device status is changed using a switch.

2. India being a tropical place we need a fan to circulate the air. While functions for a thermostat control have been provided, similar functionality for a fan is missing. Functionality expected: (a) Switch On/OFF fan, (b) Increase/Decrease speed, (c) Set speed to 1(2,3,4,5) (just like a fan speed controller)

[SinricSwitch.ino.txt](https://github.com/kakopappa/sinric/files/2067576/SinricSwitch.ino.txt)

kakopappa, 2018-06-05:
@AshokBaijal Thank you for your feedback. 

1. I think you have to query the device information from the api and check the status.
https://github.com/kakopappa/sinric/wiki/API-Documentation

2. Right now, Amazon does not have a speed controller. Closest thing you get is the thermostat controller
https://developer.amazon.com/docs/device-apis/alexa-thermostatcontroller.html

AshokBaijal, 2018-06-06:
Attaching a new .ino file. Two changes have been made
1. Switch status is saved in EEPROM and is recovered on power-on so that the status of switches is restored after power failure to their old state. (Required in places where power supply is unreliable)
2. A delay of 3 seconds between touch-switch toggle has been built in to take care of the issue raised by Jury1985. (Settable by changing DEBOUNCE_DELAY)
3. After power is restored a delay of 30 seconds has been added to take care of the time taken by the router to boot/stabilise. (Commented out in the attached file)
[SinricSwitch.ino.txt](https://github.com/kakopappa/sinric/files/2075341/SinricSwitch.ino.txt)

closed 2018-06-25 by AshokBaijal

Talha909, 2019-01-26:
Please help me. I want to add EEPRom in the script. I am new in this. 

`
int device1 = 23;
int device2 = 22;
int device3 = 19;
int device4 = 18;
int device5 = 5;
int device6 = 17;
int device7 = 16;
int device8 = 4;
 
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <WebSocketsClient.h> //   get it from https://github.com/Links2004/arduinoWebSockets/releases
#include <ArduinoJson.h> // get it   from https://arduinojson.org/ or install via Arduino library manager
#include <StreamString.h>
 
WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;
 
#define MyApiKey   "34af406e-6244-4663-997c-34b5f56ddb8a" // TODO: Change to your   sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "Home" // TODO:   Change to your Wifi network SSID
#define MyWifiPassword   "53883098" // TODO: Change to your Wifi network password
 
#define HEARTBEAT_INTERVAL 300000 // 5   Minutes
 
uint64_t heartbeatTimestamp = 0;
bool isConnected = false;
 
void setPowerStateOnServer(String   deviceId, String value);
void setTargetTemperatureOnServer(String   deviceId, String value, String scale);
 
// deviceId is the ID assgined to your   smart-home-device in sinric.com dashboard. Copy it from dashboard and paste   it here
 
void turnOn(String deviceId) {
if   (deviceId == "5c35ed05010d8c32b5565580") // Device ID of device 1
{
Serial.print("Turnevice id: ");
Serial.println(deviceId);
digitalWrite(device1, LOW);
}
else if (deviceId ==   "5c35ed93010d8c32b5565588") // Device ID of device 2
{
Serial.print("Turnevice id: ");
Serial.println(deviceId);
digitalWrite(device2, LOW);
}
else if (deviceId ==   "5c35ee0e010d8c32b556558b") // Device ID of device 3
{
Serial.print("Turnevice id: ");
Serial.println(deviceId);
digitalWrite(device3, LOW);
}
else if (deviceId ==   "5c35ee59010d8c32b556558d") // Device ID of device 4
{
Serial.print("Turnevice id: ");
Serial.println(deviceId);
digitalWrite(device4, LOW);
}
else if (deviceId ==   "5c36e5b304018c4adb252073") // Device ID of device 5
{
Serial.print("Turnevice id: ");
Serial.println(deviceId);
digitalWrite(device5, LOW);
}
else if (deviceId ==   "5c36e5be04018c4adb252075") // Device ID of device 6
{
[... 203 lines cut ...]
StreamString databuf;
root.printTo(databuf);
 
webSocket.sendTXT(databuf);
}
 
//eg:   setPowerStateOnServer("deviceid", "CELSIUS",   "25.0")
void setTargetTemperatureOnServer(String   deviceId, String value, String scale) {
DynamicJsonBuffer jsonBuffer;
JsonObject& root =   jsonBuffer.createObject();
root["action"] =   "SetTargetTemperature";
root["deviceId"] = deviceId;
 
JsonObject& valueObj =   root.createNestedObject("value");
JsonObject& targetSetpoint =   valueObj.createNestedObject("targetSetpoint");
targetSetpoint["value"] = value;
targetSetpoint["scale"] = scale;
 
StreamString databuf;
root.printTo(databuf);
 
webSocket.sendTXT(databuf);
}

`