kakopappa/sinric #70: How can I use sinric with servos?

oscarlloydjohn, 2018-06-15:
I've been looking to make some smart automated blinds as a free time project and have been investigating different ways to do it. I have a LoLin nodeMCU and have decided that Sinric looks easiest to implement as I am a beginner to Arduino (I have some basic knowledge though). MQTT could work also but that is a pain to connect to Alexa which is my main goal.

I have managed to get communication between Alexa and my nodeMCU working well, I can see the communication messages on the serial monitor. I can't seem to get my servo to move though. Here is a copy of my sketch:

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <Servo.h>
#include <WebSocketsClient.h> //  get it from https://github.com/Links2004/arduinoWebSockets/releases 
#include <ArduinoJson.h> // get it from https://arduinojson.org/ or install via Arduino library manager
#include <StreamString.h>

Servo servo1;
ESP8266WiFiMulti WiFiMulti;
WebSocketsClient webSocket;
WiFiClient client;

#define MyApiKey "removed" // TODO: Change to your sinric API Key. Your API Key is displayed on sinric.com dashboard
#define MySSID "removed" // TODO: Change to your Wifi network SSID
#define MyWifiPassword "removed" // TODO: Change to your Wifi network password
#define ServoPin 3
#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 

uint64_t heartbeatTimestamp = 0;
bool isConnected = false;

void setPowerStateOnServer(String deviceId, String value);
void setTargetTemperatureOnServer(String deviceId, String value, String scale);

// deviceId is the ID assgined to your smart-home-device in sinric.com dashboard. Copy it from dashboard and paste it here

void turnOn(String deviceId) {
  if (deviceId == "removed") // Device ID of first device
  {  
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
    servo1.attach(ServoPin);
    servo1.write(180);
  } 
  else if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of second device
  { 
    Serial.print("Turn on device id: ");
    Serial.println(deviceId);
  }
  else {
    Serial.print("Turn on for unknown device id: ");
    Serial.println(deviceId);    
  }     
}

void turnOff(String deviceId) {
   if (deviceId == "removed") // Device ID of first device
   {  
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
     servo1.attach(ServoPin);
     servo1.write(0);
   }
   else if (deviceId == "5axxxxxxxxxxxxxxxxxxx") // Device ID of second device
   { 
     Serial.print("Turn off Device ID: ");
     Serial.println(deviceId);
  }
  else {
     Serial.print("Turn off for unknown device id: ");
     Serial.println(deviceId);    
  }
}

void webSocketEvent(WStype_t type, uint8_t * payload, size_t length) {
  switch(type) {
    case WStype_DISCONNECTED:
      isConnected = false;    
      Serial.printf("[WSc] Webservice disconnected from sinric.com!\n");
      break;
    case WStype_CONNECTED: {
      isConnected = true;
      Serial.printf("[WSc] Service connected to sinric.com at url: %s\n", payload);
      Serial.printf("Waiting for commands from sinric.com ...\n");        
      }
      break;
    case WStype_TEXT: {
        Serial.printf("[WSc] get text: %s\n", payload);
        // Example payloads

        // For Switch or Light device types
        // {"deviceId": xxxx, "action": "setPowerState", value: "ON"} // https://developer.amazon.com/docs/device-apis/alexa-powercontroller.html

        // For Light device type
        // Look at the light example in github
          
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject((char*)payload); 
        String deviceId = json ["deviceId"];     
        String action = json ["action"];
        
        if(action == "setPowerState") { // Switch or Light
            String value = json ["value"];
            if(value == "ON") {
                turnOn(deviceId);
            } else {
                turnOff(deviceId);
            }
        }
        else if (action == "SetTargetTemperature") {
            String deviceId = json ["deviceId"];     
            String action = json ["action"];
            String value = json ["value"];
        }
        else if (action == "test") {
            Serial.println("[WSc] received test command from sinric.com");
        }
      }
      break;
    case WStype_BIN:
      Serial.printf("[WSc] get binary length: %u\n", length);
      break;
  }
}

void setup() {
  Serial.begin(115200);
  
  WiFiMulti.addAP(MySSID, MyWifiPassword);
  Serial.println();
  Serial.print("Connecting to Wifi: ");
  Serial.println(MySSID);  

  // Waiting for Wifi connect
  while(WiFiMulti.run() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  if(WiFiMulti.run() == WL_CONNECTED) {
    Serial.println("");
    Serial.print("WiFi connected. ");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }

  // server address, port and URL
  webSocket.begin("iot.sinric.com", 80, "/");

  // event handler
  webSocket.onEvent(webSocketEvent);
  webSocket.setAuthorization("apikey", MyApiKey);
  
  // try again every 5000ms if connection has failed
  webSocket.setReconnectInterval(5000);   // If you see 'class WebSocketsClient' has no member named 'setReconnectInterval' error update arduinoWebSockets
}

void loop() {
  webSocket.loop();
  
  if(isConnected) {
      uint64_t now = millis();
      
      // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
      if((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
          heartbeatTimestamp = now;
          webSocket.sendTXT("H");          
      }
  }   
}

// If you are going to use a push button to on/off the switch manually, use this function to update the status on the server
// so it will reflect on Alexa app.
// eg: setPowerStateOnServer("deviceid", "ON")
void setPowerStateOnServer(String deviceId, String value) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["deviceId"] = deviceId;
  root["action"] = "setPowerState";
  root["value"] = value;
  StreamString databuf;
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}

//eg: setPowerStateOnServer("deviceid", "CELSIUS", "25.0")
void setTargetTemperatureOnServer(String deviceId, String value, String scale) {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["action"] = "SetTargetTemperature";
  root["deviceId"] = deviceId;
  
  JsonObject& valueObj = root.createNestedObject("value");
  JsonObject& targetSetpoint = valueObj.createNestedObject("targetSetpoint");
  targetSetpoint["value"] = value;
  targetSetpoint["scale"] = scale;
   
  StreamString databuf;
  root.printTo(databuf);
  
  webSocket.sendTXT(databuf);
}

I have defined the servo pin in the sketch and I am powering the servo (mg995) with external 5v because the board cannot seem to handle it. (Possibly my board is faulty? It seems kinda weird sometimes)

I am unsure about how to use the payload section, I think that is my problem.

Here are the messages from the serial monitor, I think they look ok:

Connecting to Wifi: removed for post
.....
WiFi connected. IP address: removed for post
[WSc] Service connected to sinric.com at url: /
Waiting for commands from sinric.com ...
[WSc] get text: {"deviceId":"removed for post","action":"setPowerState","value":"OFF"}
Turn off Device ID: removed
[WSc] get text: {"deviceId":"removed for post","action":"setPowerState","value":"ON"}
Turn on device id: removed for post

Thanks!

kakopappa, 2018-06-16:
Have you tried moving your servo in a test project?

oscarlloydjohn, 2018-06-16:
> Have you tried moving your servo in a test project?

> Is it using a common GND?

Thanks for the reply, I just tried moving the servo using the example from the video and my own and neither made the servo move.  This is the sketch I wrote:
#include <Servo.h> 
 
Servo myservo;  
 

void setup() 
{ 
  myservo.attach(2);  // attaches the servo on GIO2 to the servo object 
} 
 
void loop() 
{ 
myservo.write(180);
  } 

The servo and Arduino are not sharing a common ground, both are powered completely independently. Could that be the problem?

kakopappa, 2018-06-16:
Could be

1.PIN is wrong
2.Broken servo
3. Not enough juice to move the servo
4. Common ground

http://www.thebox.myzen.co.uk/Tutorial/Power_Supplies.html

kakopappa, 2018-06-16:
>    1. Common ground
>
> Servo spins using the test sketch now, thanks!
>
> My sinric code doesn't make it spin though. I think it is a problem with
> servo1.attach(ServoPin); being in the wrong place or something to do with
> payload.
>
> Thanks

If you have it in the setup, it should be fine.

Put all in and check whether that works. If it does then create a method
and move the code out. Should be simple from here

oscarlloydjohn, 2018-06-16:
> If you have it in the setup, it should be fine.
Put all in and check whether that works. If it does then create a method
and move the code out. Should be simple from here

I moved servo1.attach to the void setup section, that didn't seem to work. Tried with the servo1.attach(ServoPin)  and servo1.write(180) in void setup section and that doesn't work either. I think there is a problem with the board or servo now.

Thanks

kakopappa, 2018-06-18:
got it working?

oscarlloydjohn, 2018-06-19:
> got it working?

Unfortunately not, I have ordered a new board and servos as they are inexpensive. Does my code look ok with the servo1.attach(ServoPin) in void setup and servo1.write() in the void turnOn/turnOff. As I am new to programming, this is all still quite confusing for me.

Thanks

Leoz83, 2018-11-13:
Will you implement [action.devices.types.BLINDS](https://stackoverflow.com/questions/51353342/google-smart-home-command-for-blinds)?

kakopappa, 2018-11-13:
@Leoz83 sure. I will put it in the next version.

mcathcart2, 2019-02-25:
Leoz83 did you get your Sketch to work?  I've been trying to use Sinric to do the same thing (blinds+Alexa) for a month now.  I had it working using kokapapps Blink code but sometimes it wont connect. So I decided to try to use Sinric. (I don't have a Problem with saying "Alexa Turn Blinds ON" = Operate Servo, I just can't get it to work.)  Any Help would be Greatly Appreciated (kokapappa or Lenz83)

Leoz83, 2019-03-14:
> Leoz83 did you get your Sketch to work?

Hi @mcathcart2, I don't use no more Sinric and i use Google Home instead of Alexa.
After some months from #150, [something has changed in Google Actions].(https://developers.google.com/actions/smarthome/traits/openclose)
Hope @kakopappa implements it soon!

mcathcart2, 2019-03-15:
I did not. I ended up going to an LDR to operate the blinds.

kakopappa, 2019-03-15:
Ok. I will add this to the plan