kakopappa/sinric #72: How to send lights value to sinric.com [examples required]

Deimos1994, 2018-06-18:
kakopappa thanks for this project!
I use google home and I want to make this command work: "ok, google make lights brighter", but for now it works only once, every time after a command "set lights to 10 percent" (or similar), and then it stops working. I think it is because Google home doesn't have data about current state of the lights (I mean how many percent lights is on right now).

This part of the code (from examples) works... (at least GH correctly respond whether lights ON or OFF) but it only works for ON/OFF, and I think we need something similar for percentage values.
```cpp
   void setPowerStateOnServer(String deviceId, String value) {
     DynamicJsonBuffer jsonBuffer;
     JsonObject& root = jsonBuffer.createObject();
     root["deviceId"] = deviceId;
     root["action"] = "setPowerState";
     root["value"] = value;
     StreamString databuf;
     root.printTo(databuf);
     webSocket.sendTXT(databuf);
    }
```

kakopappa, 2018-06-18:
Thanks for reporting this issue. I will check and get back to you on this.

Question: After sometime, does Google Home commands stop working? for an example Once i linked the action all worked fine. After like 24 hours Google Home says cannot connect to sinric. When I unlink and link it starts to work again. Did you happen to notice this?

Deimos1994, 2018-06-18:
@kakopappa It works pretty stably for me. Yes, it said a couple of times "I can't reach sinric", but then I asked one more time and it worked. It is a really rare event, at least for me. Almost always it works fine.
I also tried disconnecting the internet and connecting it back (just to test it) it works fine, like nothing happened, works very reliably. (I use 1-minute heartbeat).

The only issue is when I ask "how bright are the main lights right now?" (or similar) it answers: "Sorry, I am unable to reach the Main lights right now. Please try again." But, if I say "Is light on?" it reports the current state correctly. (I used the code above to send current state to Sinric)

Deimos1994, 2018-06-18:
@kakopappa Yes, now I have that issue that you ware telling me about, it says **"It looks like Sinric is unavailable right now.**". When I ask it to sync devices it says **"An error occurred while syncing Sinric."**
And yes, when I unlink and link account it starts to work again.

kakopappa, 2018-06-18:
I have updated the server with this change.

1. When you set the brightness, Sinric will save it to the local database

2. When google request it via QUERY, it will report with the saved values

3. Here is the function if you want your ESP client to set the current brightness on server

void setBrightnessOnServer(String deviceId, int value) {
     DynamicJsonBuffer jsonBuffer;
     JsonObject& root = jsonBuffer.createObject();
     root["deviceId"] = deviceId;
     root["action"] = "SetBrightness";
     root["value"] = value;
     StreamString databuf;
     root.printTo(databuf);
     webSocket.sendTXT(databuf);
    }

Deimos1994, 2018-06-18:
@kakopappa, thanks for your work. I've tryed the code above, but now it always says that light is off, even if I ask about brightness. (everything else works as before).

The code above sends this to Sinric:
`{"deviceId":"5b2664ef2b9f2034XXXXXXXX","action":"SetBrightness","value":100}`
but GH still thinks that lights is off, it should say on, or 100% on.
(I send new payload to sinric after every change of brightness).
Maybe I am doing something wrong, anyways thanks for help.

kakopappa, 2018-06-19:
sorry about breaking stuff. I will test this properly and get back to you.

kakopappa, 2018-06-23:
1. I found the issue related to the setting the device status off and fixed it. Sinric was resetting on/off status when you set the brightness level. the underlying issue was a big one. Thanks 

2.  Regarding google home "It looks like Sinric is unavailable right now." issue. Someone else just created an issue for this. We will discuss this issue there.
https://github.com/kakopappa/sinric/issues/79

please close the issue when you have tried

Deimos1994, 2018-06-23:
@kakopappa, yes, it works now, thank you very much.

closed 2018-06-23 by Deimos1994

Hansta98, 2020-01-25:
Hey guys, could you please help me on
I am pretty a newbie in sinric. I just found it yesterday evening but I am still interested to discover more. The only things I want programm are two Alexa-controlled LED-strips (turn them on/off + set there brightness over my voice (turn on/off just works with the switch_example)). I just read your chat above but I have no plan what to change: Generally how to combine more diverse functions in one scretch (where I have to tell the programm which device get appealed)
The other extention would be: I have a DHT22 which gives me two values (temperature + humidity). I want realize that when I ask Alexa "What is the temperature/humidity" that she tells me. Everything should run on a Wemos D1 mini in combination with a Echo Dot 3the generation.
Thank you in the meantime guys 👍 
Ceep on!

sivar2311, 2020-01-26:
The future of Sinric is [SinricPro](https://sinric.pro).
We have the examples for your needs here:

- [Light](https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/examples/Light/Light.ino)
- [Temperaturesensor](https://github.com/sinricpro/esp8266-esp32-sdk/blob/master/examples/temperaturesensor/temperaturesensor.ino)

Temeprature example can be used directly, because it's based on DHT22.
For the light project, you need to implement the code for controlling your LED stripes.

harymk, 2020-03-08:
kakopappa brightness not updating on server.

void setBrightnessOnServer(String deviceId, int value) {
DynamicJsonDocument root(1024);
Serial.println(value);
root["deviceId"] = deviceId;
root["action"] = "SetBrightness";
root["value"] = value;
StreamString databuf;
serializeJson(root, databuf);
webSocket.sendTXT(databuf);
}