kakopappa/sinric #73: Is it possible to check temperature inside (from a sensor) using Google Home?

Deimos1994, 2018-06-18:
It would be nice to have something like **“hey google, what is the temperature in my room?”**
I have many different sensors connected to my ESP8266, like humidity, pressure, voltage, and 5 temperature sensors, it would be very nice to ask google home about current readings from those sensors.
Probably Google home doesn't support it yet, so maybe I should not open this issue.
But maybe there is a way to use the thermostat in sinric and use it as a temperature sensor.

I added a thermostat called "sensor" and assigned it to a room where Google home is in, now when I ask **"what is the temperature inside"** it answers : **"Sorry, I am unable to reach the thing and the sensor right now. Please try again."** when I ask it there no activity via WebSockets, no payloads at all.

kakopappa, 2018-06-18:
Thanks. This seems doable with temperaturesetting. 
https://developers.google.com/actions/smarthome/traits/temperaturesetting#response

I will implement this and let you know.

kakopappa, 2018-06-23:
Implemented and tested. Should work with the new version update.

kakopappa, 2018-06-23:
example commands I used:

set the thermostats to 24
whats the temperature inside

All temperatures are in Celcius

When you turn on the thermostat: by default it will be set to cool and 23 'C

Deimos1994, 2018-06-23:
@kakopappa I send this so Sinric: 
`{"action":"SetTargetTemperature","deviceId":"5b2e307ed4e1da66de26d75e","value":{"targetSetpoint":{"value":"CELSIUS","scale":"23.8"}}}`
Is it correct ?
I say : **what temperature inside**, google says: **Sorry, I am unable to reach the sensor right now. Please try again.**

kakopappa, 2018-06-23:
AFK

Code looks good. What happens when you ask GH to set the temperature and
ask the temperature?

Does it work for you?

Deimos1994, 2018-06-23:
@kakopappa 
when I say: **set thermostat to 29** and then I ask: what is the temperature inside it says **It's currently 29 degrees.** (it works).
But I still can not set the temperature from my ESP8266, I send that JSON payload every minute with the current temperature, but GH still says It's currently 29 degrees.

Deimos1994, 2018-06-23:
@kakopappa I also recieve this payload from Sinric when I say when I say: **set thermostat to 29** `{"deviceId":"5b2e307ed4e1da66de26d75e","action":"action.devices.commands.ThermostatTemperatureSetpoint","value":{"thermostatTemperatureSetpoint":29}}`

kakopappa, 2018-06-23:
I think values you are sending wrong. 

`"value":{"targetSetpoint":{"value":"CELSIUS","scale":"23.8"}}}`

it should be

"value":{"targetSetpoint":{"value":"23.8","scale":"CELSIUS"}}}

Deimos1994, 2018-06-23:
@kakopappa, Ok it works now. 
Please fix typo in the example here: https://github.com/kakopappa/sinric/blob/master/arduino_examples/thermostat_example.ino

`//eg: setPowerStateOnServer("deviceid", "25.0", "CELSIUS")`
should be:
`//eg: setTargetTemperatureOnServer("deviceid", "25.0", "CELSIUS")`

closed 2018-06-23 by Deimos1994

Deimos1994, 2018-06-24:
@kakopappa is there a simple way to send humidity to Sinric?
GH thinks there is a humidity sensor in my room, because when I ask: **what is the humidity inside my room**, GH responds with: **The Sensor inside shows the humidity is 23 percent.**
(I've named my thermostat "Sensor inside")
This is not a high priority, but it would be a nice thing to have. Thanks for your work.

frappesc, 2018-12-14:
@Deimos1994 it is possible to have a code example? i'm trying to read the temp from a DHT11
Thanks

maciak, 2019-04-04:
Hi,
About this question with humidity. Is there any way to get it? I mean it's already sent to google home in Thermostat dht example.

pedrofnm, 2020-05-03:
Hello, im making this project. My goal is use a dht11 ESP 01 module to check temperature/humidity in my house rooms with google home using sinric. The problem is that i cant make it work. I ve tried the code above but didnt work because(i think) im not able to link it to device ID. I have created a action in google actions and it shows on GH the thermostat device. But without any data. I guess the problem is in the  code as it dont have a place to put sinric device id. Can you guys help me with this? Im trying to find a tutorial for this kinda of project but i think there isnt any. If you need more info just let me know.

pedrofnm, 2020-05-04:
OK nvm i got it working, i searched other topics you guys made and i found a working code. At least it shows the real temperature now:) I wonder if its possible to check humidity now?