kakopappa/sinric #75: Please add Power Level Controller capability to Thermostat device

AshokBaijal, 2018-06-20:
Though Percentage Controller capability is available on device type 'Thermostat' please add Power Level Controller (https://developer.amazon.com/docs/device-apis/alexa-powerlevelcontroller.html) capability to the device so that we can make a FAN controller. Though the percentage controller capability is nearly similar to the PowerLevelController capability, the verbal command of powerlevelcontroller capability is more appropriate.

kakopappa, 2018-06-20:
Sure. I will update this issue when it’s ready. Likely on weeken

AshokBaijal, 2018-06-20:
Thanks Kakopappa. I have tried the Thermostat and as stated in your sample code the response is that the device does not respond. Alexa responds when we set the temperature to x value but most of the time the Alexa App hangs after any change is made to the Thermostat. Is it because the Amazon server is not getting updated because of a lack of response packet. Though I dont know much but if you see there is only one response jason for thermostat at https://developer.amazon.com/docs/device-apis/alexa-thermostatcontroller.html#response which may need to be updated with the current temperature settings. Please consider it. The app hangs when I try to switch off the thermostat which means any change in values causes the app to crash.

kakopappa, 2018-07-02:
@AshokBaijal can you check and close this issue please?

AshokBaijal, 2018-07-04:
Thanks. Will check and report this weekend for any problem. Closing issue.

closed 2018-07-04 by AshokBaijal