kakopappa/sinric #76: Please add Channel Controller capability to TV device

AshokBaijal, 2018-06-20:
Though I have not yet tried the Sinric TV device, after looking at your github page (https://github.com/kakopappa/sinric/wiki/Supported-devices-and-their-capabilities.) the 'ChannelControllerCapability' appears to be missing. Is there a miss on the github page referred above or is the capability still to be added. Please refer to https://developer.amazon.com/blogs/alexa/post/4982d143-7f95-4d02-b259-3f39cb34fd4a/how-to-control-your-entertainment-devices-with-the-smart-home-skill-api where Amazon talks of 6 capabilities for entertainment devices.

AshokBaijal, 2018-06-22:
Checked the TV setup today. The Channel Controller capability is available though this is not listed on the Wiki.
[SinricTV Test.txt](https://github.com/kakopappa/sinric/files/2127670/SinricTV.Test.txt)
[SinricTV Test.txt](https://github.com/kakopappa/sinric/files/2127671/SinricTV.Test.txt)

 I tested the responses and summary results are as under: (Detailed test report attached)
1. Power Controller - OK
2. Channel Controller
    (a) SkipChannel always changes by 1 even though instructed to skip by, say 5, The return value of channelCount is always 1 or -1
3. Input Controller
    (a) Though the return value shows the input like HDMI 1 etc, the response by Alexa is "Sorry, TV is not responding"
4. Playback Controller
    (a) Only "Play", "Pause", "Stop" commands are supported, the other commands like "Next", "Previous", "Fastforward", "Rewind", " StartOver" are not supported. The "supportedOperations"  value during Discovery probably only include ["Play", "Pause", "Stop"]. You may consider changing the endpoint capability set during discovery to include the other supported operations.
5. Speaker  - OK

kakopappa, 2018-06-23:
Channel Controller
(a) I looked at the code. I am only forwarding Amazon Alexa response to ESP module.  Even though the documentation says 

```
"payload": {
      "channelCount" : 5
 }

```

if you are getting 1 when you request for 5, then likely Amazon is sending 1

or it could be a problem with the response. 

```
{
        "namespace": "Alexa.ChannelController",
        "name": "channel",
        "value": {
          "number": "1234",
          "callSign": "callsign1",
          "affiliateCallSign": "callsign2"
        },
        "timeOfSample": "2017-02-03T16:20:50.52Z",
        "uncertaintyInMilliseconds": 0
      }

```
https://developer.amazon.com/docs/device-apis/alexa-channelcontroller.html#changechannel

the problem here was I had the "number": "1234" hardcoded in the response. I changed the code to save the channel when you "ChangeChannel" and use the "ChangeChannel"'s number to calculate the new channel number when you send "SkipChannel".

Input Controller
(a) I checked the code, my response back to Alexa seems correct here. I am not sure why "Sorry, TV is not responding" Check whether the command is allowed in your locale.

https://developer.amazon.com/docs/device-apis/alexa-inputcontroller.html

Playback Controller
(a) - To be honest I never thought anyone would want to use this, So I did not add it at that time.

I have updated it now.

AshokBaijal, 2018-06-23:
Thanks for your prompt response. Will test further as I progress with an Alexa controlled IR controller for my home theater setup.

I find Sinric a stable platform for my ESP devices as I continue to add Alexa control to my home.

closed 2018-06-23 by AshokBaijal