kakopappa/sinric #80: can't get "White" color with echo dot

pico563, 2018-06-28:
Hi 
I can't get the "White" color with Alexa also not all colors are available,  I am comparing with my magic home light strip.
is  there a colors table that Alexa can recognize ? 
also can we get the color name from json as string value.
thanks.

mmphego, 2018-06-29:
Sorry to digress, but what app is that?

pico563, 2018-06-29:
the app. is Blynk .

Deimos1994, 2018-06-30:
I have the same problem (or similar) with my google home.
Payloads from Sinric:
for Cyan `{"deviceId":"5b2664ef2b9f20348a7be984","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"cyan","spectrumRGB":65535}}}`

for Orange `{"deviceId":"5b2664ef2b9f20348a7be984","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"orange","spectrumRGB":16744192}}}`

(All above works fine.)

But here is for White:
`{"deviceId":"5b2664ef2b9f20348a7be984","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"white","temperature":5000}}}`
As you can see it receives "temperature" instead of "spectrumRGB".

I of course can make some workaround on my side, but I think we need to set properties of the light on Sinric.com, because now Sinric or GH thinks that I can set the colour temperature of my lights, which is not true, I can only set a color.

pico563, 2018-06-30:
Hello Deimos
I am working mainly on echo dot I didn't try GH yet.
till now I found more than 30 colors Alexa can respond to by voice commands.
my problem is that Alexa don't show color name as GH did.
may be  Kakopappa  have an answer for this.

pico563, 2018-06-30:
colors with Alexa  are not 100% compatible with  GH. strange?

{"deviceId":"xxxx","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"sky blue","spectrumRGB":8900331}}}
{"deviceId":"xxxx","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"violet","spectrumRGB":8323327}}}
{"deviceId":"xxxx","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"purple","spectrumRGB":8388736}}}
{"deviceId":"xxxx","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"royal blue","spectrumRGB":9062}}}
 {"deviceId":"xxxx","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"lavender","spectrumRGB":11894492}}}

GH
 {"deviceId":"xxxx","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"blue violet","spectrumRGB":9055202}}}
 {"deviceId":"xxxx","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"dark violet","spectrumRGB":9699539}}}

Alexa
{"deviceId":"xxxx","action":"SetColor","value":{"hue":282,"saturation":1,"brightness":0.8275}}{"deviceId":"xxxx","action":"SetColor","value":{"hue":271,"saturation":0.8097,"brightness":0.8863}}

**_only last 2 color  give same RGB result  the rest are different._**

Deimos1994, 2018-07-01:
@pico563 Hi, I don't have Alexa so I thought it would be the same. I think Sinric.com redirects parts of JSON responses from GH servers and Amazon Alexa servers to ours ESP8266, and since GH and Alexa made by two different companies they represent color differently. Seems like Alexa uses HSV (HSB) color model. But GH, on the other hand, uses HEX colour as Int (by default), buy in addition it also sends the name of the color. More about GH here:
https://developers.google.com/actions/smarthome/traits/colorspectrum

Sinric.com doesn't make conversion "color name" to "color value", it is made by GH or Alexa on their servers. 
Sinric just get something similar to this:
 `{"deviceId":"xxxx","action":"SetColor","value":{"hue":282,"saturation":1,"brightness":0.8275}}`
or this:
 `{"deviceId":"xxxx","action":"action.devices.commands.ColorAbsolute","value":{"color":{"name":"dark violet","spectrumRGB":9699539}}}` and then it sends it to us.
On the link above about GH it says: **colorModel Can be set to string hsv to indicate device preference for the HSV (hue, saturation, value) color model. Default is rgb.**
And maybe there is something similar for Alexa, and if so then you'll get RGB instead HSV (if implemented by Sinric).
But if Alexa don't understand "set lights to white" then nothing can be done here.

P.S. anyways, I am not an expert in this stuff, it is just my thoughts.

pico563, 2018-07-01:
Hi 
I still don't know how to get color name from Amazon Echo

**_in picture E is for Echo Dot and G for GH_**

[image][](url)

closed 2018-07-03 by pico563