kakopappa/sinric #81: Issue with "play" command - TV

AshokBaijal, 2018-06-29:
I have made an ESP8266 IR controller for my Philips Soundbar. While most of the commands are working Alexa is finding it a problem to respond to the "PLAY" command as the command gets diverted to the Music App which comes with Alexa and starts playing music by some group called Soundbars. This is an Amazon issue. I am trying to contact them. Just keeping the community in the loop and if someone can suggest a solution.

Attaching the code for the IR controller
[SinricSoundbar.ino.txt](https://github.com/kakopappa/sinric/files/2147617/SinricSoundbar.ino.txt)

mmphego, 2018-06-29:
Dude Note that your api key is on the file...
Serious security issue

AshokBaijal, 2018-06-29:
Sorry buddy, oversight. Code deleted.

AshokBaijal, 2018-06-29:
Got a call from Amazon Support. They have suggested to change the device name from Soundbar to something else. Let me experiment further.

AshokBaijal, 2018-06-29:
Tried after changing the device name. However, Alexa responds to the Play command by invoking the Music Skill. 

This is an Amazon issue and not Sinric issue. As such closing the issue here.

closed 2018-06-29 by AshokBaijal