kakopappa/sinric #86: can't set thermostat Ambient Temperature on server.

pico563, 2018-07-05:
Hi
thermostatTemperatureSetpoint----->     **"work fine"**.
//eg: setTargetTemperatureOnServer("deviceid", "25.0", "CELSIUS")

but I don't know how to do it for the rest
thermostatTemperatureAmbient------------->| Current observe temperature, in C
thermostatTemperatureSetpointHigh-------->| Current high point if in heatcool mode, for a range
thermostatTemperatureSetpointLow--------->| Current low point if in heatcool mode, for a range
thermostatHumidityAmbient----------------->| Float. 0-100. If available from the device.

thanks

kakopappa, 2018-07-06:
@pico563 I will update the server code and update this issue when it's ready. (likely on this weekend)

pico563, 2018-08-07:
@kakopappa 
Hello 
When this will be available on Sinric

thermostatTemperatureAmbient------------->| Current observe temperature, in C
thermostatTemperatureSetpointHigh-------->| Current high point if in heatcool mode, for a range
thermostatTemperatureSetpointLow--------->| Current low point if in heatcool mode, for a range
thermostatHumidityAmbient----------------->| Float. 0-100. If available from the device.

For now I am using   "thermostatTemperatureSetpoint" with "Ambient"
But I don't know what is the best speed to push data on server 
So there is no load for Sinric to write on server

thanks

emnik, 2018-11-01:
Hi!
How do you actually send the current temperature to the server?

The setTargetTemperatureOnServer("deviceid", "25.0", "CELSIUS") sends the TARGET temperature not the current...

thanks

kakopappa, 2018-11-02:
Hi, 

There are two parts to this. 

Request:
When you send SetTargetTemperature or AdjustTargetTemperature, the server will save value and scale in the local db for this device.

1. If you set the temperature using [SetTargetTemperature](https://developer.amazon.com/docs/device-apis/alexa-thermostatcontroller.html#settargettemperature). The server will get the targetSetpoint temperature and save it in the local db. 

2. If you adjust using AdjustTargetTemperature the server will adjust the previously saved value with the new value.

Response:
According to https://developer.amazon.com/docs/device-apis/alexa-thermostatcontroller.html#response This is the response format

{
      "namespace": "Alexa.ThermostatController",
      "name": "targetSetpoint",
      "value": {
        "value": 25,
        "scale": "CELSIUS"
	}
}

the server will put the saved value in the database and send it back to the Alexa.