kakopappa/sinric #88: Android app

kakopappa, 2018-07-10:
Guys, I have released the Android app to Google Play. 

https://play.google.com/store/apps/details?id=com.sinric

Please install it and let know how it goes

mmphego, 2018-07-10:
Nice one, will you also add the src code to the repo?

Deimos1994, 2018-07-10:
@kakopappa I've installed the app... it shows devices that I previously added at sinric.com.
I can't control my lights right away because of different payloads representation. (In other words, I have code on my esp8266 that designed to work with GH, not with the app).
When I want to turn lights on
via GH I get this:
`{"deviceId":"5b2664ef2b9f20348a7be984","action":"action.devices.commands.OnOff","value":{"on":true}}`
via Sinric app I get this:
`{"deviceId":"5b2664ef2b9f20348a7be984","action":"setPowerState","value":"ON"}`
And if I would have Alexa I think I would have a third payload representation.

My thoughts: maybe we need some unified way to represent all of those commands. Or maybe we need some correct code on ours esp8266/esp32 that can correctly parse all those JSON payloads that we get from sinric.com

kakopappa, 2018-07-10:
@Deimos1994 Yes I thought about it alot. 

Option 1: Use unified command set
Option 2: Use Alexa command set
Option 3: Use Goggle command set

I decided to go with Option 2. Because most of the users are using Amazon Alexa. For them, app will work right after the installtion without any modifications to ESP code.

After reading your comments, i think there is option 4

Option 4: Send Alexa, Google Home both commands.

What do you guys think ?

AshokBaijal, 2018-07-11:
Nice. The app looks exciting and what was needed. Some suggestions:
1. I have Sinric devices spread across rooms. Can you build an option to group the icons room wise and have separate screens for different rooms.
2. Right now we can switch on or off a device from the app. Later can you build options for operating the different capabilities of the devices.

al8943, 2018-07-16:
Great start @kakopappa as regards to Option 4 I would prefer a setting in the app as I have set my devices to respond to either.

kakopappa, 2018-07-21:
sorry about the late response. 

I have updated the Android app with Google / Alexa command format now.

In Menu -> Settings -> Now you can select the command format between Google Home / Alexa.

Please update the app and give it a try

al8943, 2018-07-21:
The app is sending different messages the google home or Alexa.

Alexa, on
[WSc] get text: {"deviceId":"5bxxxxxxxxxxxxxxxxxxxxxx","action":"setPowerState","value":"ON"}
App set to Alexa, on
[WSc] get text: {"deviceId":"5bxxxxxxxxxxxxxxxxxxxxxx","action":"powerState","value":"ON"}

Alexa, off
[WSc] get text: {"deviceId":"5bxxxxxxxxxxxxxxxxxxxxxx","action":"setPowerState","value":"OFF"}
App set to Alexa, off
[WSc] get text: {"deviceId":"5bxxxxxxxxxxxxxxxxxxxxxx","action":"powerState","value":"OFF"}

Google, on
[WSc] get text: {"deviceId":"5bxxxxxxxxxxxxxxxxxxxxxx","action":"action.devices.commands.OnOff","value":{"on":true}}
App set to Google, on
[WSc] get text: {"deviceId":"5bxxxxxxxxxxxxxxxxxxxxxx","action":"action.devices.commands.OnOff","value":"{\"on\":true}"}

Google, off
[WSc] get text: {"deviceId":"5bxxxxxxxxxxxxxxxxxxxxxx","action":"action.devices.commands.OnOff","value":{"on":false}}
App set to Google, off
[WSc] get text: {"deviceId":"5bxxxxxxxxxxxxxxxxxxxxxx","action":"action.devices.commands.OnOff","value":"{\"on\":off}"}

kakopappa, 2018-07-21:
@al8943 Thanks a lot man. 

I fixed the command formats accordingly. I have fixed these issues in app  (v3.0) . It will be available on Google Play to update shortly.

al8943, 2018-07-21:
Thanks, works perfectly now.

digiulio81, 2018-10-23:
Congratulations for the work done. I am using SINRIC and everything is ok. I wanted to ask you if the APP is updated in real time on the status of the switches / lights.
I have modified your code to make sure that every 5 seconds check if the status of the switch has changed, only if it has changed is updated on the server. But I saw that it does not change the switch state on the APP.

kakopappa, 2018-10-23:
Thanks.

Yes it’s updated real-time. Check the code to send the on/off status. You
can find the example at the bottom of the page

digiulio81, 2018-10-24:
OK, I re- check it and post it in a new issue. 
The code is for a pulse switch, there is also a current probe to check if the light is on or off. 
tnx

digiulio81, 2018-10-24:
#define HEARTBEAT_INTERVAL 300000 // 5 Minutes 
#define REFRESH_INTERVAL   10000   // 10sec					// the time we need to wait to check if the light is on or off

uint64_t previousMillis = 0; 							                  // millis() returns an unsigned long.
uint64_t currentMillis = 0;
uint64_t heartbeatTimestamp = 0;

bool isConnected = false;
bool isON = false;
bool Prev_isON = false;

void loop() {
  webSocket.loop();

  if (isConnected) {
    uint64_t now = millis();

    // Send heartbeat in order to avoid disconnections during ISP resetting IPs over night. Thanks @MacSass
    if ((now - heartbeatTimestamp) > HEARTBEAT_INTERVAL) {
      heartbeatTimestamp = now;
      webSocket.sendTXT("H");
    }
  }
  

  uint64_t currentMillis = millis(); 			// grab current time
  // check if "REFRESH_INTERVAL" time has passed 
  
  if ((unsigned long)(currentMillis - previousMillis) >= REFRESH_INTERVAL) {
    currentMeter();
    delay(500);

    if (isON == LOW) {
      if (Prev_isON == HIGH) {
        Serial.println("Light switched off manualy");
        setPowerStateOnServer("deviceid", "OFF");
      }
    }

    if (isON == HIGH) {
      if (Prev_isON == LOW) {
        Serial.println("Light switched on manualy");
        setPowerStateOnServer("deviceid", "ON");
      }
    }

    Prev_isON = isON; // remember prev state for next pass
    // save the "current" time
    previousMillis = currentMillis;
  }
}

On serial monitor works good, I don't understand!