kakopappa/sinric #89: Problem with alexa skill connecting to sinric

noobipaul, 2018-07-10:
Hi @kakopappa . I have been using Alexa controlled esp8266 switches via sinric for a while now without any problems.
This evening they seemed to have stopped.  I disabled the sinric Amazon skill, then tried to enable it again but get the error shown in attached screen shot? Any ideas what the problem is? Thanks.
[image]

kakopappa, 2018-07-11:
Hello. Can you try now? Let me know if you still have this problem

noobipaul, 2018-07-11:
Thank you. Working now :)

closed 2018-07-11 by noobipaul

minimatte77, 2018-12-04:
Hi @kakopappa . I have been using Alexa controlled esp8266 switches via sinric for a while now without any problems.
This evening they seemed to have stopped. I disabled the sinric Amazon skill, then tried to enable it again but get the error shown in attached screen shot? Any ideas what the problem is? Thanks.

I have the same problem, what can it depend on?
Thank you