kakopappa/sinric #90: TV skill notification problem

akshay8284, 2018-07-14:
You had done a very well job there thank you.
i would like to know that if sinric is open source?

T.v skills for volume are working perfectly good. 
T.v skill for channels carries serial data, but Alexa responds "Sorry, TV isn't responding." why is this happening only while changing channel but works good for volume commands.

kakopappa, 2018-07-21:
@akshay8284 

I would like to know that if sinric is open source? 

Not yet. I am worried about someone finding a bug in the code and ruining this for everyone. May be one day.

T.v skill for channels carries serial data, but Alexa responds "Sorry, TV isn't responding." 

Can you give me more details about the command you use? Like exact phrase you used?

amitabh-srivastav, 2018-11-26:
@kakopappa , Sir you have done excellent work, but  when you ask ALexa:
"Alexa, change channel to National Geographic on Tata Sky" or
"Alexa, change channel to 128 on Tata Sky", we get the response from Alexa
 that **"Sorry, TV isn't responding."**
Although, the set top box correctly sets the channel, but the response is very disappointing.
It is humbly requested to please look in to matter and resolve, as response to other commands are OK.